randdusing/cordova-plugin-bluetoothle #406: Maximum Stack Size Exceeded

unclespode, 2017-03-07:
On some devices, when uploading a firmware file I'm getting a Maximum Stack Size Exceeded error on bytesToEncodedString.

I found the following, which has the exact function you're using and other people are complaining of the same problem. This only affects some devices (in my case an iPhone 5). The suggested looping method in the article fixes the problem. 

http://stackoverflow.com/questions/9267899/arraybuffer-to-base64-encoded-string

randdusing, 2017-03-09:
How large is your firmware?

Can you see if below works for you:
```
btoa([].reduce.call(new Uint8Array(bufferArray),function(p,c){return p+String.fromCharCode(c)},''))
```

If it works, you can just use that instead of using the provided helper function until I update the plugin.

unclespode, 2017-03-09:
Only 200k or so IIRC. On all my android devices it's been fine - only an issue on iPhone.

I'll try this tomorrow. I used the loop as suggested and it seemed to solve the problem then I was having other issues, so I can't be 100% certain it's a working solution. I'll try yours.

randdusing, 2017-03-13:
I copied my from the StackOverflow post as well. Were you able to verify it?

unclespode, 2017-03-14:
As far as I can tell, yes - that fixes the issue. However, I'm now having another issue that I'll keep here because it's related and might affect other people having similar problems. Can always pull it out to a separate issue if deemed so.

So I'm trying to write (without response) about 100k. This is for DFU. I subscribe for notifications, and then I'm using the writeQ function to write the data.

On Android, it's working fine and I get a notification every 400 bytes as it uploads, but finally getting a notification of success.

On iOS, it's giving me the same notifications every 400 bytes, but I just stop getting them at anywhere from 6400 - 8800 bytes. Doesn't seem to stop at a consistent point. It'll just hang there forever. 

If I try it with write instead of writeQ it doesn't seem to work at all. 

Any thoughts?

unclespode, 2017-03-14:
Just to add to this - my working (and now possibly proven) theory is that it's writing data quicker than my device can handle so it's losing many of the chunks.

I edited the iOS file and added to the end of the do/while loop:

[NSThread sleepForTimeInterval:.0.005];

With this, it seems to be writing correctly but behavior isn't quite as expected. It seems to be writing, then all of a sudden I get all the notifications for the bytes being received all at once. 

Do we perhaps need an extra parameter for writeQ where you can add a manual delay for devices we know can't handle it?

randdusing, 2017-03-18:
How many bytes are you writing per writeQ call? You should still split your data up into chunks. And the device itself should be sending acknowledgements every chunk.  Like you may call writeQ with 100 bytes, wait for an acknowledgement, call writeQ again, etc.

unclespode, 2017-03-21:
I'm just sending the whole lot down. I thought the point with writeQ is that it does the queue-ing for you. If that is a valid technique - surely should be taken care of at your level, and just have a chunkSize option?

unclespode, 2017-03-22:
I've set up chunking, with an optional delay between chunks. I should add I'm using noResponse (as required by the DFU) which may be the cause of the issue.

With 128 bytes, with 0 delay, it doesn't receive all the data. It's like it starts sending the next chunk before it's received the previous one.

Through trial and error, I found 32 byte chunk size, with 16ms delay between calls was stable. What I don't understand is why Android has no issues?

randdusing, 2017-03-29:
writeQ was mostly intended to minimize the performance degradation caused by the Javascript to Cordova bridge. Instead of crossing that bridge five times for 100 bytes, you could cross it once. It wasn't intended to write a whole file at once since the assumption is that the Javascript layer would wait for some sort of acknowledgement from the BLE device before sending more bytes.

I can't remember why I chose 20 bytes. I may have thought it was an upper limit for write with no response. Ideally it should be configurable from the Javascript layer if it can be higher than 20 bytes. I think this value will also depend on phone, OS, BLE device and the BLE connection parameters.

Android probably has no issue because I implemented my own queue on top of it. The write won't be called until the previous write is completed. iOS implements their own queue and making hundreds of calls in a row is probably causing it to stop working.

unclespode, 2017-03-29:
On Android, I was sending it to writeQ in one chunk and it was working.

On iOS, doing that caused, well packet loss essentially.

As I'm using noResponse, I can't really write a queue, because I have no way of knowing when the bytes are written. Through experimentation I have hit a byte size / delay that works - but ultimately, it's just guess work and feels a little hacky.

So what's happening in your code? is iOS using it's own queue, or is your queue in place? Could we make iOS use the same (optional?) queue as Android?

randdusing, 2017-03-30:
iOS uses its own queue and you're likely running into some sort of overflow without the delay. No callback is returned on iOS with write without response, so I can't implement a queue in the same fashion as Android.