randdusing/cordova-plugin-bluetoothle #407: Service not showing up

aCollins2101, 2017-03-15:
I'm trying to setup a connection between two devices and the devices services aren't being advertised in scans. The advertising device is a Samsung Galaxy S5. The callback says that services are being added and advertising started. Removal of services and stopping advertising also seems to work without issue. The other weird thing is that the advertising is not being detected at all by the sample app linked in the readme so I have to use an third party app called BluetoothLEScanner. I'm essentially using the code from the readme but, I thought I'd post the relevant sections.

The code I'm using to add a service:
```
var serviceParams = {
    service: "1234",
    characteristics: [
    {
        uuid: "ABCD",
        permissions: {
            read: true,
            write: true,
            //readEncryptionRequired: true,
            //writeEncryptionRequired: true,
        },
        properties: {
            read: true,
            writeWithoutResponse: true,
            write: true,
            notify: true,
            indicate: true,
            //authenticatedSignedWrites: true,
            //notifyEncryptionRequired: true,
            //indicateEncryptionRequired: true,
        }
    }
    ]
};
bluetoothle.addService(bluetoothModel.callbackSuccess.serviceAdded,
    function (res) {
        Logger.log("Failed to add serice: " + JSON.stringify(res), Logger.labels.BLUETOOTH, Logger.priority.ERROR);
    },
    serviceParams);
```

In case it's relevant I'm advertising with:
```
var params = {
    "services":["1234"], //iOS
    "service":"1234", //Android
    "name":"Hello World"
};
bluetoothle.startAdvertising(
    function (result) {
        Logger.log("Advertising Started with result: " + JSON.stringify(result),
            Logger.labels.BLUETOOTH,
            Logger.priority.DEBUG);},
    function (result) {
        Logger.log("Advertising Failed with result: " + JSON.stringify(result),
            Logger.labels.BLUETOOTH,
            Logger.priority.ERROR)
    },
    params);
```

randdusing, 2017-03-16:
So even the third party scanner isn't detect your device?

aCollins2101, 2017-03-16:
The third party app can detect my device when I start advertising but it does not see the service that I added. It says there are no services detected for my phone.

randdusing, 2017-03-18:
Do you receive any error when calling addService? And are you waiting until addService resolves before calling startAdvertising? Try out the example in the Angular wrapper and see if the file transfer example works on your device.

aCollins2101, 2017-04-02:
Thanks for the reference to Angular example looking at it solved my issue and helped me progress a lot faster through the rest of my code. It turns out I was badly referencing services when scanning for them. Thanks very much.

closed 2017-04-02 by aCollins2101