randdusing/cordova-plugin-bluetoothle #409: How to receive subscription, if working with multiple HTML-documents

ebeling, 2017-04-06:
In my project I have multiple HTML-sites. One site is for showing bluetooth- and connection-state, another site should show diagrams with received values of bluetooth-device. If the connection-sequence on the first site is successful I like to switch to the second site (there are more sites than 2). After switching I can use functions like bluetoothle.isInitialized, bluetoothle.isEnabled,  bluetoothle.isScanning and so on and I can store the parameter of the connected device in the sessionStorage, before switching. But what is the right way to receive the subscribeResults on the second site? Must I call bluetoothle.subscribe once again?

randdusing, 2017-04-06:
What do you mean by multiple HTML sites? An app usually has one "site". You most likely don't want to connect to the same device multiple times from different sites.... assuming the device would even allow that.

ebeling, 2017-04-06:
Correct is that I don't want to connect the device multiple times. But I've designed Website with Navigation and if I click on another navigation-item a new html- site will be called. (first site is index.html, second site is diagram-site.html for example).  The Cordova-App has one webview, but it don't must be single html-site?!

randdusing, 2017-04-07:
This more of a SPA design question. Check out the example in the angular wrapper for an example

ebeling, 2017-04-07:
Does it mean that I only can use your plugin in single page application architecture? The angular wrapper example is this 
https://github.com/randdusing/ng-cordova-bluetoothle
?

randdusing, 2017-04-07:
I'm sure you can but I've never done, so I can't offer much advice. I would
highly recommend some sort of SPA framework

ebeling, 2017-04-07:
Thank you for your answer! I've tried something but it doesn't work. I've never used SPA-framework. I think, I have to learn much more ...