randdusing/cordova-plugin-bluetoothle #410: Scan On Android no devices found

NadirBertolasi, 2017-04-08:
Hi,
I am trying to scan for devices on a Galaxy s8 with Android 7.0

var params = {
      "services": null,
      "allowDuplicates": true,
      "scanMode": bluetoothle.SCAN_MODE_LOW_LATENCY,
      "matchMode": bluetoothle.MATCH_MODE_AGGRESSIVE,
      "matchNum": bluetoothle.MATCH_NUM_MAX_ADVERTISEMENT,
      "callbackType": bluetoothle.CALLBACK_TYPE_ALL_MATCHES,
    };

    bluetoothle.startScan(function(res){
      console.log(res)
    }, function(err){
      console.log(err)
    }, params);

I get the success fuction called onlly when iOS devices are found. I cannot get any Android device (with visibility= true).

randdusing, 2017-04-10:
Last I checked Android 7.0 broke scanning. Does this work on Android 6 or 7.1?

NadirBertolasi, 2017-04-12:
I have just tested Scan on Android 7.0 and 5.1.  Both devices callback success if on detection of bluetooth on iPhone or Mac. But they cannot see each others. It seems like there is a problem in scan and discover Android devices.

randdusing, 2017-04-14:
Are you expecting your Android devices to return in the scan results? Are they advertising as BLE devices?

NadirBertolasi, 2017-04-15:
They have both bluetooth and visibility on. Have I to set them up in advertise mode? How can I do it?
Thanks

randdusing, 2017-04-15:
You need an app that specifically advertises the device using Bluetooth LE (not Bluetooth Classic!). You could make one using this plugin using the Angular wrapper example pretty easily.

NadirBertolasi, 2017-04-15:
OK thank you, I thought the behavior was similar to iPhone. I will certainly follow the example to advertise through my device.

randdusing, 2017-06-29:
Closing due to inactivity. Please reopen if you have any followup.

closed 2017-06-29 by randdusing