randdusing/cordova-plugin-bluetoothle #411: cordova-plugin-bluetoothle and Ionic 2

ebeling, 2017-04-11:
Your example is running with Ionic 1. Is your cordova-plugin-bluetoothle also usable in Ionic 2? I'm not familiar with Ionic and I have to decide, what I have to learn. Many experienced developers recommend Ionic 2.

randdusing, 2017-04-11:
The wrapper only supports Angular 1 and thus Ionic 1. But it should be pretty simple to convert to Angular 2 if you're familiar with Angular 2.

sublimator, 2017-08-21:
@randdusing 

I started on an ionic-native wrapper here:
https://gist.github.com/sublimator/6969bdf0457f7ac809fa90ba600990ee

It "works", though I've only covered the api surface that we needed so far. 

Also, thanks for the lib btw. I was getting hair-pulling mystery bugs using the `@ionic-native/ble` plugin, then switched to this and started smiling again.

eiannone, 2017-12-24:
I'm trying this, too. I've found useful information also here: https://github.com/ionic-team/ionic-native/blob/master/DEVELOPER.md

somq, 2018-01-06:
@eiannone,
I had the same issue and needed a convenient install method.
So I wrapped and [published to npm](https://www.npmjs.com/package/ionic-plugin-bluetoothle) somewhat what @sublimator posted. Here is the [source](https://github.com/somq/ionic-plugin-bluetoothle-dev). Feel free to request a PR if you wish to.

eiannone, 2018-02-03:
Thank you @somq . 
I'm a novice on Angular/Ionic so maybe I'm wrong, but I was wondering if would be better to set the _initialize_ method to return an Observable, in place of a Promise, given that the callback is supposed to be called continuously (https://github.com/randdusing/cordova-plugin-bluetoothle#initialize).

How about defining it the following way?

```Javascript
  @Cordova({callbackOrder: 'reverse', observable: true})
  initialize(params?: {request?: boolean, statusReceiver?: boolean, restoreKey?: string}):
  Observable<{ status: 'enabled' | 'disabled', message? : string }> { return; }
```

somq, 2018-02-03:
Hi @eiannone,

Yes, you are right, it should be an observable.
What the [initialize method](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/654786b40aa08cfa4560795774f123b754373a23/src/android/BluetoothLePlugin.java#L941) does is to check if the bluetooth state is on (button toggled On on the phone topbar) and if not, force an enable or request the user to do it.

I've done this part in a perm service using [diagnostic plugin](https://github.com/dpa99c/cordova-diagnostic-plugin/blob/master/src/android/Diagnostic.java#L579) which explains why it wasn't necessary for me to correct this method to return an observable instead of a promise.

Anyway, it should be modified, there is still a lot of methods missing, or missing/misdefined types. I will take a look at it to finish it properly (including a proper doc) when I have some spare time but I can't tell when.

If you feel it, you are free to pull a request with the modifications.

eiannone, 2018-02-09:
Before submitting a PR I'm testing it, and it seems that even when _initialize_ is defined as an Observable, the subscribed callback is called only on the first time, but not when bluetooth status changes.
I'm calling it like this:

```Javascript
this.ble.initialize({request: true, statusReceiver: true, restoreKey: "CBT-Diagnostic"})
    .subscribe((res) => {
        // This code should be called every time bluetooth is enabled or disabled, but it doesn't
        alert('ble: ' + res.status);
    });
``` 

Do you have any advice about what could be possibly wrong?

ebeling, 2018-06-13:
Thank you @randdusing @somq and @eiannone! I've tried the ionic-native-wrapper and also the initialize-method as observable. I have debugged this in Android and realized, that the params are empty in BluetootheLePlugin.java - initialize - method. Therefore no receiver for bluetooth-changed action will be registered. The problem I think is, that the "initialize"-method doesn't has an error-callback and the "@Cordova callbackOrder"-argument expects success and error callbacks as the first two arguments (documented https://github.com/ionic-team/ionic-native/blob/master/DEVELOPER.md ). I have tried to use the "@Cordova successIndex" - argument, but there isn't a "argumentsIndex" - argument and I don't know, how to tell ionic-native the argument-list of the initialize-method of cordova-bluetootle-plugin?!

ebeling, 2018-06-13:
As workaround I changed the parameter list in cordova-plugin-bluetoothle in file "www/bluetoothle.js" from
```javascript
initialize: function(successCallback, params) {
    cordova.exec(successCallback, errorCallback, bluetoothleName, "initialize", [params]);
 }
``` 
to
```javascript
initialize: function(successCallback, errorCallback, params) {
          cordova.exec(successCallback, errorCallback, bluetoothleName, "initialize", [params]);
 }
```
Now every bluetooth-changed event is triggered.