randdusing/cordova-plugin-bluetoothle #413: Closing the app leaves "stuck" connections on Android.

mebrunet, 2017-04-13:
I'm noticing that on many Android devices (a variety of models and APK version), when I close the app, it doesn't terminate all the BLE connections I had initiated. As a result, of few of the connections get "stuck", and they can only be terminated by toggling the Bluetooth on/off. 

It's as though the OS doesn't fully garbage collect and release the resources when the app is no longer there to reference them. No amount of "force quitting", or closing all background apps seems to release them either.

It's not consistent... the same phone, running the same code, trying to connect to the same set of device will sometimes hold a few connections,  and sometimes release them.

Any thoughts on how this can be addressed? It's a big issue for us.

mebrunet, 2017-04-13:
Simple way to recreate the bug:

## Setup
1) Create a new phonegap project
2) Modify `config.xml` to include this plugin from npm
3) Modify `index.html` to load script `js/bleTest.js` with contents:
```
var bleTest = {
    // replace MAC_X and NAME_X accordingly
    testDevices: [
        'MAC_1': 'NAME_1',
        ....
        'MAC_6': 'NAME_6', 
    },
    run: function () {
        bluetoothle.initialize(console.log, { restoreKey: 'cordovabletest' });
        Object.keys(this.testDevices).map((key) => {
            console.log('Attempting to connect: ' + this.testDevices[key]);
            bluetoothle.connect(console.log, console.error, { address: key });
    },
}
```
4) Build with PGB and install on an Android mobile device
5) Open the app with the mobile device in debug mode and USB connected to a computer, then use `chrome://inspect` to attach to the phonegap webview and get a remote console. 
6) Execute `bleTest.run()` in the console. (Make sure Bluetooth and Location services are on first)
7) After connections' success/error callbacks all fire, exit the app using Android tab/process manager.

## Result: 
On some devices connections get "stuck", and can only be terminated by toggling the Bluetooth on/off.

If you're struggling to recreate the issue, close the app with the tab manager before all the connection callbacks have fired... this seems to re-create the issue across a larger set of devices.

randdusing, 2017-04-14:
I don't have much time to devote to this project anymore, so I'm not sure when I'd be able to make the change. The coding part is simple, but testing on a wide variety of devices is very time consuming. PRs are welcome for the changes outlined below.

The CordovaPlugin class has an onDestroy method where cleanup can be done. You should be able to iterate through each of the devices and call disconnect or close on them. I would start with close and see if that's good enough. If that's not good enough, call disconnect and then close.

randdusing, 2017-06-29:
Assuming I get to this, how are you verifying whether the device is actually connected? Does your BLE device have some sort of indication?

mebrunet, 2017-07-02:
Yes our BLE device has an LED to indicate its connection state.

But it turns out the bug is deeper than your plug-in. Using a native
Android app we could recreate the bug quite easily across a number of
phones when following the recommended connect/disconnect process. It would
appear that it is actually related to BLE driver implementation.

We've mitigated the problem by having the peripheral disconnect when it no
longer receives a sort of "keep alive".

randdusing, 2017-07-04:
Ahh, Android and Bluetooth is fun to work on. I'm going to close this unless there's something that can be added at the plugin level to get around it, but it doesn't sound like that's possible...

closed 2017-07-04 by randdusing

xixiaxixi, 2019-12-03:
Recently I met this bug, too. Maybe I know the reason of this bug, but I am not able to solve it.

Reason I guessed:
The `Map connections` saves the information, but it would be deleted every time the app stops.
After restarting, the `connections` is re-initilized as a `new Map<>()`. The refrence is lost and the app can never release the previous connection.

Solution: 
release the connection when onDestroy?
or save the instance(gatt1 below or `connection` in source code)?

Example: 
```
BluetoothGatt gatt1 =  bluetoothAdapter.getRemoteDevice("D7:7F:6E:B1:6A:75").connectGatt(cordova.getActivity().getApplicationContext(), false, bluetoothGattCallback);
BluetoothGatt gatt2 =  bluetoothAdapter.getRemoteDevice("D7:7F:6E:B1:6A:75").connectGatt(cordova.getActivity().getApplicationContext(), false, bluetoothGattCallback);
gatt1.close();  # connction is not release fully
gatt2.close();
```

The ble connection doesn't release after I call `navigator.app.exitApp()`, What's worse, the connection doesn't release after I restart the app. 
My module stops broadcasting when it is connected, so I can never scan the ble module unless I kill the app. But, in the console, I can connect the ble module again and close. The `close` works normally, but the connection still doesn't release. (forgive my poor English:)
 @randdusing

randdusing, 2019-12-16:
@xixiaxixi  Thanks for the details. I thought restarting the app (even without disconnecting or closing) generally closed out any connections. Is the device you're connecting to your own custom device? I'm wondering if this is an issue with a particular peripheral device, or possibly a specific Android device. I'm not sure what the plugin can do to release connections besides calling #close() anyways. #disconnect() followed by #close() wouldn't be possible because the callback wouldn't be fired before the app dies.