randdusing/cordova-plugin-bluetoothle #414: isEnabled false before initialization

sunny-mittal, 2017-04-15:
Many (most) methods return "bluetooth not initialized" or equivalent if called before `initialize`, but `isEnabled` returns false until initialization occurs, making it hard to make conditional decisions based on the bluetooth radio state. Now I don't know enough about this level vis a vis app development, so is this a limitation of the OS or something that can be changed/modified in the plugin? Thanks!

randdusing, 2017-04-15:
You need to call bluetoothle.initialize() before anything else

sunny-mittal, 2017-04-15:
In that case, shouldn't it return the same error as other calls (i.e. Bluetooth not initialized) rather than `false`? I think it'd be more consistent and less confusing :D Thanks for the quick response, though!

randdusing, 2017-04-16:
Oh I misread your comment. Yah perhaps isEnabled and isScanning should return an error, but true/false is simpler to handle.

sunny-mittal, 2017-04-16:
That's fair. I'm closing this issue as resolved. Hopefully someone else will benefit from it and thanks again for the explanation!

closed 2017-04-16 by sunny-mittal