randdusing/cordova-plugin-bluetoothle #417: Possible to speed up disconnect detection on Android?

ELundby45, 2017-04-25:
I've noticed on Android I don't get a notification that the connection has been lost for 20 seconds. On iOS I'm getting the event within just a couple of seconds.

Is there any way (beyond adding some sort of heartbeat/keepalive) to get Android closer to the iOS experience?

randdusing, 2017-04-26:
There's another issue logged where someone tried some other ideas to "predict" a disconnect by monitoring the rssi, but I believe it only worked on some Android phones / versions.

ELundby45, 2017-04-26:
Thanks for your response. I've checked out the referenced [issue](https://github.com/randdusing/cordova-plugin-bluetoothle/issues/236). It [appears](https://issuetracker.google.com/issues/37074926) the Android team won't be updating the connection supervisor timeout anytime soon (which is hard coded to 20s). I'll have to find another solution. Thank you @randdusing for this great library!

closed 2017-04-26 by ELundby45