randdusing/cordova-plugin-bluetoothle #418: Using writeQ()

NXTAdmin, 2017-04-27:
1.       What is the max size, number of bytes, that can be sent to the writeQ() function?  
2.       Is there any way to determine if writeQ() is busy or has sent out all bytes?
3)	What happens if writeQ is called with additional data before the data from the previous call is sent out?

randdusing, 2017-04-28:
1. On Android, it should theoretically have no limit. The queue is within the plugin and not limited to a specific length.
On iOS, the limit is most likely the underlying limit of the internal queue. I'm not sure what this is and I imagine it changes based on iOS version and device hardware.
2. There's not a method, but it could be added on Android. On iOS, the queue is handled internally, so the plugin has no way of knowing.
3. On Android, it will break. Only one queued write at a time. This could be changed to be per device and even per characteristic. On iOS, it should be fine.

randdusing, 2017-06-29:
Closing due to inactivity. Please reopen if you have any followup.

closed 2017-06-29 by randdusing