randdusing/cordova-plugin-bluetoothle #419: Receiving Rx data using bluetoothle.subscribe() - packets out of order

NXTAdmin, 2017-04-27:
When we use the subscribe() method and start receiving a lot of data the packets will occasionally arrive out of order.  On Android  the out of order packet, 1 20-byte chunk, seems to loosely correlate with when a Tx packet is sent from the phone to our hardware.   Using the BT sniffer we can see that the hardware is sending the packets in the correct order.  Have you seen out of order packets being received and is there anything we can do about it?

randdusing, 2017-04-28:
I haven't seen that before, but I haven't really been looking. Unfortunately I don't think much can be done at the plugin level because it doesn't really receive much per subscription call besides the actual data.

alexhisen, 2017-05-01:
If you are using Cordova-Android 6.x platform, you might want to try downgrading to Cordova-Android 5.x, which uses a different communication mechanism between native and JavaScript.

randdusing, 2017-06-29:
Closing this since I don't think there's anything that can be done at the plugin level, but I'm curious if you found any resolution.

NXTAdmin, 2017-08-02:
Since we control both ends of the pipe we ended up modifying the way packets are sent to the phone by adding a 1 byte header which is a packet counter.  On the phone we keep an array of packets and reorder based on the 1 byte packet counter before passing the data along to the next level.

timburke, 2019-01-07:
If anyone is still hitting this issue on this plugin, I have finally tracked down what it is.  The short story is that Cordova does not actually guarantee that Plugin results will come back from native code to javascript in the order they were sent.  See https://github.com/don/cordova-plugin-ble-central/issues/625#issuecomment-451788534 for a more complete description of exactly how this can happen.

The upside is that this issue should be entirely fixable within this plugin itself by adding a monotonic sequence number to each notification and manually reordering them on the javascript side before sending them to the user callbacks.

timburke, 2019-01-07:
@randdusing In case you are interested, here is a reference implementation of a minimal fix that could be ported to resolve this issue.  It modifies the callback process to include a monotonically increasing sequence number with each notification.  The code was written for `cordova-plugin-ble-central` so I'm not sure how easily it would be to port over to this plugin.

The java code implementing it is here:
https://github.com/iotile/cordova-plugin-ble-central/commit/d7f55548da85b601a7bc68fb768d8290c1659da3

A corresponding typescript class that reads the sequence numbers and dispatches notification callbacks in order is here:

```typescript

/**
 * A sliding window reordering helper class.
 * 
 * This class takes tuples (value, sequence_number) and a callback that should be
 * called as `callback(value)` but in order of sequence_number.  If a sequence 
 * number is received out of order, it puts it into a holding list to be dispatched
 * at the appropriate time. 
 */

export class SlidingWindowReorderer<T> {
    private nextExpected: number;
    private onHold: {[key: number]: T};
    private callback: (value: T) => void | Promise<void>;

    constructor(callback: (value: T) => void | Promise<void>) {
        this.callback = callback;
        this.onHold = {};
        this.nextExpected = 0;
    }

    public call(value: T, sequence: number | null) {
        /**
         * If there is not a sequence number present, just pass the callback through
         * without reordering it.
         */
        if (sequence == null) {
            this.callback(value);
            return;  
        }

        if (sequence != this.nextExpected) console.warn(`Received out of order: expected ${this.nextExpected} got ${sequence}`);

        this.onHold[sequence] = value;

        this.tryDispatchInOrder();
    }

    private tryDispatchInOrder() {
        while (this.nextExpected in this.onHold) {
            try {
                let value = this.onHold[this.nextExpected];
                delete this.onHold[this.nextExpected];

                this.nextExpected += 1;

                this.callback(value);
            } catch (err) {
                console.error("Error in callback in SlidingWindowReorderer", err);
            }
        }
    }
}
```

alexhisen, 2019-01-07:
@timburke Are you using Cordova-Android 6+? Have you tried doing window.cordova.exec.setNativeToJsBridgeMode(window.cordova.exec.nativeToJsModes.ONLINE_EVENT);
(i.e. instead of using the new default window.cordova.exec.nativeToJsModes.EVAL_BRIDGE mode)?

timburke, 2019-01-07:
@alexhisen I have not tried it but based on the code paths involved I don't believe that would work.  The particular line that causes the issue is here:
https://github.com/apache/cordova-android/blob/master/framework/src/org/apache/cordova/CordovaBridge.java#L62

It allows queued callbacks to be sent immediately to javascript rather than waiting in the queue if they happen to be queued right when a call to `exec` on any plugin happens from javascript.  As I read the code it looks like that shortcut path is active for all bridge modes.

A way to fix it would be to set `DISABLE_EXEC_CHAINING` to stop that behavior:
https://github.com/apache/cordova-android/blob/master/framework/src/org/apache/cordova/CordovaBridge.java#L61

but that seems like it would have a significant performance hit since it would remove that fast-path entirely for any plugin results.

alexhisen, 2019-01-07:
@timburke I believe that whole bridge mechanism is new to 6 and ONLINE_EVENT is the older mechanism that doesn't use that code. When we first tried 6 a couple of years ago, it completely destroyed our BLE transfers with massive quantities of out of sequence packets. Executing the above line in deviceReady brought it down to an occasional and manageable level. And performance is at least no worse than on iOS.

timburke, 2019-01-07:
@alexhisen Understood.  Thanks for the additional context there.  It's good to know there are other potential workarounds as well.  

As an FYI, we have implemented the sequence numbers with automatic reordering now internally and it completely resolved the problem in our testing so far.  We see about 2-4% of out-of-order notifications in our use case that are automatically fixed before handing off to our javascript callback.

butcher-07, 2019-05-21:
Hey guys, any plans on integrating this functionality? Just asking, in case anybody already did it...

randdusing, 2019-05-26:
I don't think it's possible to do this at the plugin level. Something like this will need to be done at the application level and coordinated between the server and client.

panuhorsmalahti, 2019-12-17:
@randdusing can you explain why application-level sequence number is needed, since @timburke posted an example fix for the issue? Bluetooth itself has sequence numbering, so it would seem unnecessary to add another application-level sequence numbering.

randdusing, 2019-12-24:
I must have misread (or not read) some of the comments above. It's look like the issue is the plugin receives everything in order, but the bridge between the plugin and JavaScript has a small chance at delivering packets out of order. Thus the plugin should send an initial sequence number value if every message. This seems pretty straightforward, but not sure how soon I can get to this. PRs welcome :)

PR #687 (2021-06-08): Fix out-of-order notifications on Android
This pull request ensures the order of ble notifications delivered to subscribers matches the order messages were received over-the-air.

This PR is based off the PR made for cordova-plugin-ble-central (https://github.com/don/cordova-plugin-ble-central/pull/656) and the implementation suggested by Tim Burke (https://github.com/randdusing/cordova-plugin-bluetoothle/issues/419#issuecomment-452058869)

Closes #419
closed 2021-06-08 by randdusing