randdusing/cordova-plugin-bluetoothle #420: background tasks

fjlogrono, 2017-05-04:
First, thank you very much for the plugin.
I have a problem, I've been trying to do something and I do not know how to solve it.
I need to get the heart rate, and if it is less than or greater than a range of values, send an alert using pushplugin.

I am also using the backgroundmode plugin.

When it enters the background, it only works correctly 30 seconds. Then the app closes. How can I continue to obtain the cardiac frequency obtained with the subscribed method and continue to send alerts if it goes out of range?

If I have bluetooth-central in the config.xml, should not be in the background running the subscribed method without stopping?

In the initilized method, I also have the restorekey.

I'm using ng-cordova-bluetooth for angulajs, and the Central Life Cycle.

thank you very much

fjlogrono, 2017-05-04:
Maybe, the subscribed method is running in the background, and I need to activate the background mode when the heart rate is out of range, send the alert, and send it back to the background. I understand that the subscribed method will continue to process the data.

I'll try that.

Android works perfectly

fjlogrono, 2017-05-04:
I think the subscribed method is not working in the background, because Ios debugger closes when a few seconds passes in the background. This happens to me even with and without the background mode plugin.

fjlogrono, 2017-05-04:
I'm starting to think that there is a memory problem, and the app is closed.

fjlogrono, 2017-05-04:
I left the application in the background and it did not close. 

Then, I have reactivated and received bluetooth data (bluetoothle.subscribe (subscribeSuccess, subscribeError, params);), and after a few seconds the application has been closed when in the background:

Terminated due to signal 9

some help?
Thank you!

fjlogrono, 2017-05-04:
This happens to me in iOS 10.

fjlogrono, 2017-05-05:
Seems to be a performance issue for my app, not its plugin. sorry for the inconvenience.

closed 2017-05-05 by fjlogrono

randdusing, 2017-05-27:
Thanks for the update and hope you figure out the performance issue.