randdusing/cordova-plugin-bluetoothle #421: Android auto-reconnect after unsubscribe, disconnect and close.

DaniloCouto, 2017-05-04:
I'm with the following situation.
I bond with an slave, then I connect with it, without autoConnect parameter.
(So the default value of would be false)
Then I subscribe within two distinct services and characteristics, so I can write freely and receive some notifications from the slave.

So, when the user hit the disconnect button, I unsubscribe to these services and caracteristics, check if the connection is up, if it is connected I disconnect and then close the connection.

But, the android tries to reconnect instantly after all this disconnect command.
So what can cause this auto-reconnect in this situation?

randdusing, 2017-05-27:
Can you list the workflow of all the calls you make? It should never reconnect automatically unless you opt in. Are you using reconnect by any chance? That could possibly cause the issue.

DaniloCouto, 2017-05-31:
Oh, that should be it.
I try to connect to an device, if by any chance it go wrong, i try to reconnect.
Maybe this happens when I connect by Reconnect.
Going to try to change it and edit here.

Thanks for the help.

randdusing, 2017-06-29:
Sorry, I never followed up on this. Can you list out the calls you make? I can try to recreate with one of my devices.

DaniloCouto, 2017-06-29:
That's fine.
I use [ngCordovaBluetoothle](https://github.com/randdusing/ng-cordova-bluetoothle.
) because I'm developing on Ionic 1.

I retry to connect at the maximum of 5 times if the connection doesn't work.
So I use this function, 

```
function tryToConnect(slave) {
        var defer = $q.defer();
        $cordovaBluetoothLE.connect({ address: slave.address, timeout: 10000 }).then(function(obj) {
            console.warn('Notify:', obj);
        }, function(objError) {
            console.error('connect', objError)
            if (objError.error == "connect") {
                $cordovaBluetoothLE.disconnect({ address: slave.address }).then(function(success) {
                    $cordovaBluetoothLE.close({ address: slave.address }).then(function() {
                        defer.reject('retry');
                    }, function() {
                        defer.reject('retry');
                    })
                }, function(objError) {
                    if (objError.error === "isDisconnected") {
                        $cordovaBluetoothLE.close({ address: dispositivo.address }).then(function() {
                            defer.reject('retry')
                        }, function() {
                            defer.reject('retry');
                        })
                    } else {
                        defer.reject('retry');
                    }
                });
            } else {
                defer.reject('retry');
            }
        }, function(success) {
            if (success.status == "connected") {
                defer.resolve();
            }
        });
        return defer.promise;
    }
```

On disconnect I basically check if the device is connected to the slave, then I unsubscribe to any services that may be connected,  then I disconnect Then I close the connection

```
$cordovaBluetoothLE.isConnected({ address: slave.address }).then(function(success) {
      if (obj.isConnected) {
           unsubscribe(slave).then(function() {
               $cordovaBluetoothLE.disconnect({ address: slave.address }).then(function(obj) {
                     $cordovaBluetoothLE.close({ address: slave.address }).then(function(obj) {
                           q.resolve();
                      }, function(obj) {
                           q.reject();
                      });
                }, function(obj) {
                    $cordovaBluetoothLE.close({ address: slave.address }).then(function(obj) {
                          q.resolve();
                     }, function(obj) {
                          q.reject();
                     });
                });
         })
```

In your bluetooth experiencie, the slave device can force an reconnect if the device is bonded?

randdusing, 2017-07-04:
I haven't done much with bonded devices lately, so I'm not sure if that may be causing the reconnect. I think you could clean up your connection logic a bit with promise chains. Basically if any error occurs, always unsubscribe, disconnect, close in your finally. You can see some examples here: https://github.com/randdusing/ng-cordova-bluetoothle/blob/master/example/index.js#L1086