randdusing/cordova-plugin-bluetoothle #422: Obtain device adapter address and name plus useful state attributes

silviucm, 2017-05-08:
Hi @randdusing ,

I was wondering whether you would accept a pull request (for Android only, I do not have Objective C experience) for a "getAdapterInfo" method, that can be invoked even before bluetoothle.initialize 

  /**
  * Retrieves a minimal set of adapter details 
  * (address, name, initialized state, enabled state, scanning state, discoverable state)
  */
  private void getAdapterInfoAction(CallbackContext callbackContext)

The method would (crucially) retrieve the address and the BLE name on the device itself.
If **bluetoothAdapter** is null, a temporary **bluetoothAdapterTmp** would be used to extract the name and the address, for zero impact to the legacy flow.

**Usage inside Cordova:**
`
 bluetoothle.getAdapterInfo(function (adapterInfo) {
          // log the result, etc JSON.stringify(adapterInfo));
          // etc etc
});
`
**Sample uninitialized JSON result:**

{"address":"18:1E:B0:D2:B2:3C","name":"SGH-M919V","isInitialized":false,"isEnabled":false,"isScanning":false,"isDiscoverable":false}

**Initialized JSON result:**

{"address":"18:1E:B0:D2:B2:3C","name":"SGH-M919V","isInitialized":true,"isEnabled":true,"isScanning":false,"isDiscoverable":false}

Obviously, if you cannot accept Android-only for sake of symmetry, I completely understand.

Regards

randdusing, 2017-05-10:
Most of those flags would be false if not initialized. But it should be
fine other than that. I'm curious why you wouldn't just called initialize.

silviucm, 2017-05-10:
My main drive was to get the address and the name. I believe only **initializePeripheral** returns the address (please correct me if I'm wrong). 
The (optional) name might not even be exposed at all.
I have an Android app pet project where there is a device details page where I intended to dump the name and the address of the device in non-peripheral mode.

Exposing the other states came to me as a secondary thought. Rather than going through a sequential callback hell to get various possible states, I thought it would be nice to have a one-off snapshot with the basic adapter details and the most recent states, and take whatever action in its callback.

randdusing, 2017-05-27:
I think getAdapterInfo that just returns the address and name is a good idea. Not sure if there's much benefit in returning the is* flags. You'll want to verify whether getAdapterInfo works when BLE is disabled.

silviucm, 2017-05-27:
Cool. I placed the method before bluetoothle.initialize and it did not explode. 
It returned isEnabled: false
I am going to test it with an explicit bluetoothle.disable, then running it to see what happens

Allow me to try selling you on the idea of is* flags one final time. 

In a scenario where the BLE adapter was suddenly disabled, or for whatever reason the UI has (somehow) lost the correct state one could be running, say:
    `bluetoothle.stopScan(stopScanSuccess, stopScanError);`
     
and end up with stopScanError callback reporting errors such as: 
    `{"error":"stopScan","message":"Not scanning"}` 

To put the state back in order, the proper (non-hacky) way would be:
invoke isInitialized: 
  in the callback invoke isEnabled if need be
    in the subsequent callback potentially invoke isScanning if need be
      in the callback potentially invoke (a potential new functionality I'm tempted to work on) isDiscoverable

I think that getting those several states all at once would dramatically reduce the callback hell

The main net new code chunk is here:
https://github.com/silviucm/cordova-plugin-bluetoothle/blob/master/src/android/BluetoothLePlugin.java#L1019

Have a look at it, let me know your feedback, and we go from there.

randdusing, 2017-05-27:
That looks good! I have a few minor comments that I can add once you make the PR

silviucm, 2017-06-01:
Closing this after https://github.com/randdusing/cordova-plugin-bluetoothle/pull/430 got merged.
In case I get my hands on a Mac I will attempt to do an iOS equivalent

closed 2017-06-01 by silviucm