randdusing/cordova-plugin-bluetoothle #423: No subscription result

puneet9, 2017-05-09:
Hello Randd,

I am trying to connect ble device using your plugin and i am successful in writing values to the characteristics and i am expecting to get 50 readings per second, but when i turn on subscription to get notification the the status never changes from subscribed to subscribedResult so i cant fetch any data, can you please explain any possible reason for this, do i need to turn on any notifications on android or anything else?
Thanks
Puneet

randdusing, 2017-05-10:
To rule out any code issues, try using the Angular wrapper example. If that
doesn't work , it's more likely an issue with your BLE device or possibly
even the phone

puneet9, 2017-05-10:
Hello Randd,

I have checked my ble with other plugin BlE scan it starts giving me
notifications perfectly the only problem with that is the readings drop
after 17 sec, that is why i wanted to try your plugin,is there any possible
reason you can think of why i am not getting subscribe result,like editing
.java file?

Thanks
Puneet

randdusing, 2017-05-27:
Not sure why. You can try using the PR regarding subscriptions on Android and see if that works for you.

randdusing, 2017-06-29:
Closing due to inactivity. Please reopen if you have any followup.

closed 2017-06-29 by randdusing

chegewara, 2019-11-11:
Hi @randdusing 
i would like to open this issue. I am using ionic with angular and here is my code to register for notifications. Code seeems to work, because i can confirm that peripheral device is receiving write to 0x2902 descriptor proper value, but nothing is received when notification/indication is sent:
```

chegewara, 2019-11-11:
After further investigation i found the issue is here:
`let value = this.ble.encodedStringToBytes(data.value);`
```
Uncaught DOMException: Failed to execute 'atob' on 'Window': The string to be decoded is not correctly encoded.
    at Object.encodedStringToBytes (http://0.0.0.0:8100/plugins/cordova-plugin-bluetoothle/www/bluetoothle.js:146:16)
    at callCordovaPlugin (http://0.0.0.0:8100/vendor.js:96651:43)
    at http://0.0.0.0:8100/vendor.js:96731:20
    at cordova (http://0.0.0.0:8100/vendor.js:96962:96)
    at BluetoothLE.push../node_modules/@ionic-native/bluetooth-le/ngx/index.js.BluetoothLE.encodedStringToBytes (http://0.0.0.0:8100/vendor.js:96264:142)
    at SafeSubscriber._next (http://0.0.0.0:8100/tab3-tab3-module.js:130:34)
    at SafeSubscriber.__tryOrUnsub (http://0.0.0.0:8100/vendor.js:111763:16)
    at SafeSubscriber.next (http://0.0.0.0:8100/vendor.js:111702:22)
    at Subscriber._next (http://0.0.0.0:8100/vendor.js:111652:26)
    at Subscriber.next (http://0.0.0.0:8100/vendor.js:111629:18)
```
and this is notification when i dont try to encode it:
```
[console.log]: "value " {
[ng]   "address": "30:AE:A4:43:A1:72",
[ng]   "name": "Le Gizmeaux",
[ng]   "service": "6E400001-B5A3-F393-E0A9-E50E24DCCA9E",
[ng]   "characteristic": "6E400003-B5A3-F393-E0A9-E50E24DCCA9E",
[ng]   "status": "subscribedResult",
[ng]   "value": "eyJoZWlnaHQiOjEuOTU5OTk5LCJ0ZXN0IjpudWxsLCJ0ZXN0MiI6InNvbWV0aGluZyBsb25nIn0="
[ng] }
```

atwing, 2022-06-21:
> After further investigation i found the issue is here: `let value = this.ble.encodedStringToBytes(data.value);`
> 
> ```
> Uncaught DOMException: Failed to execute 'atob' on 'Window': The string to be decoded is not correctly encoded.
>     at Object.encodedStringToBytes (http://0.0.0.0:8100/plugins/cordova-plugin-bluetoothle/www/bluetoothle.js:146:16)
>     at callCordovaPlugin (http://0.0.0.0:8100/vendor.js:96651:43)
>     at http://0.0.0.0:8100/vendor.js:96731:20
>     at cordova (http://0.0.0.0:8100/vendor.js:96962:96)
>     at BluetoothLE.push../node_modules/@ionic-native/bluetooth-le/ngx/index.js.BluetoothLE.encodedStringToBytes (http://0.0.0.0:8100/vendor.js:96264:142)
>     at SafeSubscriber._next (http://0.0.0.0:8100/tab3-tab3-module.js:130:34)
>     at SafeSubscriber.__tryOrUnsub (http://0.0.0.0:8100/vendor.js:111763:16)
>     at SafeSubscriber.next (http://0.0.0.0:8100/vendor.js:111702:22)
>     at Subscriber._next (http://0.0.0.0:8100/vendor.js:111652:26)
>     at Subscriber.next (http://0.0.0.0:8100/vendor.js:111629:18)
> ```
> 
> and this is notification when i dont try to encode it:
> 
> ```
> [console.log]: "value " {
> [ng]   "address": "30:AE:A4:43:A1:72",
> [ng]   "name": "Le Gizmeaux",
> [ng]   "service": "6E400001-B5A3-F393-E0A9-E50E24DCCA9E",
> [ng]   "characteristic": "6E400003-B5A3-F393-E0A9-E50E24DCCA9E",
> [ng]   "status": "subscribedResult",
> [ng]   "value": "eyJoZWlnaHQiOjEuOTU5OTk5LCJ0ZXN0IjpudWxsLCJ0ZXN0MiI6InNvbWV0aGluZyBsb25nIn0="
> [ng] }
> ```
> 
> EDIT This is solution:
> 
> ```
>       if (data.value) {
>         let value = this.ble.encodedStringToBytes(data.value);
>         // let _value = bytesToString(value);
>         console.log('value ', data);
>         // this.notifyValue = _value.height;
>       }
> ```
> 
> What i found that first notification value received by ionic app from subscriber, also first device discovered with scan, is null. That seems to be issue, which can be easy fixed with workaround.

awesome, this worked for me. It would've prolly took me days to find this out. Thanks a bunch!