randdusing/cordova-plugin-bluetoothle #424: onServicesDiscovered does not get called.

t2wu, 2017-05-10:
On Android 6 & 7, after I included `cordova-plugin-firebase`, it seems like service discovery becomes unreliable. `onServicesDiscovered` doesn't get called.

Check another branch, check back, rebuild, it can work. Check another branch again, come back, rebuild, doesn't work. Same code. This is really bizarre.

t2wu, 2017-05-10:
It appears that I can get it to work reliably if after connect I wait 600 ms (as mentioned by someone from Stackoverflow) before I discover.

I will report here if there is any more issue encountered after this change.

randdusing, 2017-05-10:
That seems bizarre. The issue only appears with that plugin? Is it trying
to do anything Bluetooth related?

randdusing, 2017-06-29:
Any update on this issue? Still experiencing it? And do you know if firebase does anything Bluetooth related?

t2wu, 2017-06-29:
Thank you so much for following up the issue. You're a great plugin author!

I do not believe Firebase has anything Bluetooth related. I just added the timeout and that was it.

However, there may be a separate issue, I do frequently run into cases on HTC (Android 6) where onServicesDiscovered just never gets returned even with the timeout. The problem never gets solve even after disabling/enabling Bluetooth and rebooting the device.

The only reliable way to resolve it is to actually download a third-party generic Bluetooth app and connect it. After that, our app would connect reliably.

randdusing, 2017-07-04:
Do you just have one device to test one? I'm curious if it happens with other devices too, or just a specific one? I have noticed that issue more on Android 7, but it seemed like something out of the plugins control.

t2wu, 2017-07-04:
With the most recent occurrence, instead of resorting to the third-party app to "reset it" so to speak, during `bluetoothle.discover` I removed `clearCache: true` and it works!

We recently started to test on more devices. We have not yet run into problems yet, but I already changed the clear cache setting.

I have two more issues. Probably neither of it is under your control, if you want me to post separate issues I'll do so.

1. On a separate issue, on one Samsung device, if connect takes more than 6 seconds (such as when device is slow to connect, or when the device isn't even around), I'll get a connect fail. HTC devices we tried and iPhone as well will never receive a return, so I set my own timeout of 30 seconds. But on Samsung that timeout is useless because failure called back is called at around 6 seconds.

2. I've got a case (LG) where I get initiate a firmware upgrade by writing on the device characteristic. The write will not get proper return (because the device switch to upgrading AP mode right after. I think it ought to return proper response to indicate success in writing. But talk to the firmware engineer about it! It is not under my control.), then, after waiting for timeout on that write, I close the device (bluetoothle.close without bluetoothle.disconnect first). But interestingly I get both success and failure callback called!

t2wu, 2017-07-18:
@randdusing BTW I was using 4.3.0. In what way does the phone/firmware issue may exhibit the difference between the third party app and the app downloaded from the app store? (I'm using [Bluetooth LE Scanner)](https://play.google.com/store/apps/details?id=uk.co.alt236.btlescan)

To answer your question, the success call back is called first, followed in split second the failure call back.

Today I run into a even stranger problem. We rented a BlackBerry Priv, using the said third party scanner I was able to detect our Bluetooth devices, but using the plugin, nothing is scanned.

t2wu, 2017-07-18:
Ah! Nevermind. I realized it is this "[Screen Overlay Detected](https://forums.crackberry.com/blackberry-priv-f440/frustrated-screen-overlay-detected-1079754/)" issue. It actually does not allow permission to be set when run into this. I have yet to solve it. For some reason that Bluetooth LE scanner permission is set to be permitted. After I removed and install that app again, I've trouble authorizing permission it myself. It's a UI issue.

randdusing, 2017-07-30:
That sounds very strange. I probably usually disabled the screen overlay while testing, but I really disliked the Facebook chat heats.

AdrGeorgescu, 2019-05-31:
I have the same issue on Android 8.1.0
The list of services is not available when I use the .discover() function.

When I use clearCache: true neither discoverSuccess or discoverError functions are called.

```
const discoverParams: BluetoothlePlugin.DiscoverParams = {
      address,
      clearCache: true
};
   
bluetoothle.discover(discoverSuccess, discoverError, discoverParams);
```

DominikTrenz, 2019-10-23:
> I have the same issue on Android 8.1.0
> The list of services is not available when I use the .discover() function.
> 
> When I use clearCache: true neither discoverSuccess or discoverError functions are called.

I got the same issue