randdusing/cordova-plugin-bluetoothle #427: Custom characteristic on Windows 10

lacos, 2017-05-20:
I try to subscribe and write a custom characteristic (service uuid: 2456e1b9-26e2-8f83-e744-f34f01e9d701 characteristic uuid: 2456e1b9-26e2-8f83-e744-f34f01e9d703) on Windows 10. 
I use the sample code and is works well on Android (I can also subscribe and write the characteristic), but on Windows 10 it doesn't.

After getting service the app displays the service uuid well, but after getting characteristics the app displays only short uuid 'e1b9' and when I try to read the characteristic it returns the following error message: Error on read: Characteristic not found.0

When I try to subscribe the characteristic (with the 128bit uuid) it returns the following error message: Error on subscribe: Invalid parameters.0

When I try to write the characteristic (with the 128bit uuid) it retuns a JSON object with status: written but the bluetooth device does not receive the package. 

What can be the problem?

The result of calling the characterisitics is the following:
```
{
  "status": "characteristics",
  "characteristics": [
    {
      "uuid": "e1b9",
      "properties": {
        "read": "true",
        "writeWithoutResponse": "true",
        "write": "true",
        "notify": "true"
      }
    },
    {
      "uuid": "e1b9",
      "properties": {
        "writeWithoutResponse": "true",
        "write": "true",
        "notify": "true"
      }
    }
  ],
  "service": "2456e1b9-26e2-8f83-e744-f34f01e9d701",
  "address": "6a376569-ba58-5368-9936-a0ed8a9803b3"
}
```

randdusing, 2017-05-27:
Characteristic 2456e1b9-26e2-8f83-e744-f34f01e9d703 doesn't exist. Characteristic e1b9 may be receiving an error since it's duplicated. What type of BLE device are you connecting to?

randdusing, 2017-06-29:
Closing due to inactivity. Please reopen if you have any followup.

closed 2017-06-29 by randdusing