randdusing/cordova-plugin-bluetoothle #432: Best practices for sending a lots of data?

hannibaldinski, 2017-06-09:
Hello,

I've to send a file which is around 42kb to a Bluetooth device. It is used to update the firmware. The device requires to receive the file in packets of 55 bytes.

What I'm doing right now is:

-> Devide the file in unit8Array's of 55 bytes 
-> Convert to encodedString using bluetoothle.bytesToEncodedString()
-> Send value using bluetoothle.write()
-> Loop till all bytes are send.

I was wondering if this would be the best approach? Or do I need to use writeQ though this one does not have any example code.

Cheers,

Julien

randdusing, 2017-06-11:
Mostly right. writeQ is really only helpful with Android since it reduces the number of times you need to talk between javascript and plugin layer. If on Android, you'll want to adjust the connection priority to higher so data can be sent quicker too. I also recommend implementing a reliable protocol on top of Bluetooth. If you just blast a ton of packets, some may be dropped or possibly arrive out of order, and then you'll have to start over.

I have a basic file transfer example in the Angular wrapper. You can search for throughputCentral/Peripheral to see how it works on both ends. You'll need two devices if you want to see it in action.

hannibaldinski, 2017-06-11:
Thanks for the suggestions! I'll have a look at the Angular wrapper and connection priority.

I've the option to write with and without response. Probably go for writing with response. Might be a bit slower, but more reliable.

randdusing, 2017-06-11:
I would do write without response. Write with response is about twice as slow if I'm remembering correctly.

hannibaldinski, 2017-06-13:
Yes, writing without response is lot slower though my current application not an issue.

closed 2017-06-13 by hannibaldinski

terraloader, 2019-08-16:
Thanks for the thread.
I had a problem with the device and did realize late that it had a maximum of 128 byte as message maximum size. I now chunk it into multiple messages and it works.