randdusing/cordova-plugin-bluetoothle #433: iOS State Preservation and Restoration

alexhisen, 2017-06-21:
# Introduction
It doesn't look like the plugin currently implements iOS State Restoration fully. It seems like iOS does Preserve the state but when the app is relaunched by iOS, the previously connected BLE devices are no longer connected and you have to re-connect to them, re-discover, etc.

I am not an iOS (or Cordova plugin) developer but I've been doing a lot of reading and experimenting on this subject and here are some thoughts on this topic.

# Background Execution
With UIBackgroundMode set to bluetooth-central (which can be done with https://github.com/randdusing/cordova-plugin-background-mode-bluetooth-central), iOS will keep the app suspended in the background and, when a connected peripheral sends a notification (or it reconnects), iOS will wake the app and it will continue execution in the background for 10 sec (from last BLE packet). State Preservation and Restoration is _not_ involved here.

If iOS has to terminate the app due to memory pressure, the BLE connection will be retained if bluetoothle.initialize() was previously called with a restoreKey. When a BLE packet comes in, iOS will actually re-launch the app into background with similar 10 sec of execution window from last BLE packet. As long as packets keep coming in, the app will keep running (and Cordova initializes normally - BTW, you can use this plugin to tell you are not in foreground - https://github.com/leomylonas/cordova-plugin-applicationstate). **Note that currently it seems to be necessary for bluetoothle.initialize() to be called with a restoreKey within 10 sec of the app launch, otherwise the BLE connection will be dropped even if BLE packets keep coming in during those 10 sec (and then the app will be suspended).**

# Restoring state
When the app is actually launched and not just awoken from a BLE trigger, the willRestoreState function in the plugin is invoked. Currently it doesn't do anything. willRestoreState receives an array of CBPeripheral objects, so it would need to re-setup them up, so that the plugin is aware the peripheral is actually connected.
https://developer.apple.com/library/content/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW11 describes the process in detail.

> It also talks about launching the app with launchOptions that contain the restoreKey, so the plugin could obtain it from there if necessary and not have to wait for JS code to call initialize. https://stackoverflow.com/questions/31000387/customizing-ioss-applicationdidfinishlaunchingwithoptions-method-in-a-cordova might be useful for this.

Currently, to JavaScript code, at least, it appears to not be (or ever was) connected and so has to be connected/discovered/subscribed. This is actually a good thing, because otherwise if the app launched and saw that it was previously connected, at least the way our app behaves, is to first close the connection just in case, which we don't want to happen. So, if the plugin were to restore the state, wasConnected() has to still 'pretend' to JavaScript that it was/is not connected, so the app can go through all the normal connect (without a disconnect/close), discover, subscribe calls to setup all the callbacks. But internally, perhaps the plugin could skip the actual BLE connect and discover processes?

Right now, presumably because it doesn't, you get two API MISUSE errors logged by CoreBluetooth:
- Forcing disconnection of unused peripheral ... Did you forget to cancel the connection?
- <CBCentralManager: 0x16ecadd0> can only accept this command while in the powered on state.

# Testing / Debugging
To simulate app termination due to memory pressure you can attach to it with Xcode and stop it or invoke some code in it that ends up calling  kill(getpid(), SIGKILL) (the hockeyApp.forceCrash() can be used for this if you use that plugin).

Note that forced user terminations of the app (from the double-tap on home button app list) are not the same and app will not be relaunched after such as termination.

You then need the still-connected peripheral to send a BLE notification for the app to be launched. In Xcode, you can see log notices from the app (including the above API MISUSE errors) under Window | Devices and then click on the tiny arrow at the bottom of the screen to Show device console.

BTW, to get your JavaScript console.log statements to show up here use this plugin - http://cordova.apache.org/docs/en/latest/reference/cordova-plugin-console/ 

# Dealing with phone reboots and device disconnections
If the peripheral goes out of range, it will not re-connect when it comes back in range if the app is terminated (it will if the app is suspended). And if the phone is rebooted or has its BLE disabled and re-enabled, peripheral will no longer be able to relaunch the app on demand. You can work around both issues through the use of the background fetch mechanism and the https://github.com/transistorsoft/cordova-plugin-background-fetch plugin.

Once that plugin is initialized (with window.BackgroundFetch.configure(callbackFn)), iOS will periodically (including after a phone reboot) launch or wake the app when it is suspended giving it 30 seconds to try to reconnect to the peripheral. The launch part will only work if the Scheme used to build the app is configured to do so. In Xcode, Product | Scheme | Edit Scheme... - under Run - Options, check the 'Launch due to a background fetch event' checkbox. After 30 seconds, the app is stopped (by default) or suspended (with stopOnTerminate: false) unless it's configured for bluetooth-central background execution and receiving BLE packets, in which case, it'll continue to execute for as long as it keeps receiving them.

When the app is awoken from suspension, callbackFn is invoked first thing. When the app is launched by a background fetch event, callbackFn is called after you call window.BackgroundFetch.configure(callbackFn). Your callbackFn must then call window.BackgroundFetch.finish() within 30 sec of app launch. In our app, we schedule the call to window.BackgroundFetch.configure in 10 sec during app startup and our callbackFn schedules the call to finish() in 5 sec. The effect of this is when the app is awoken from suspend by background fetch, it runs for 5 seconds unless the app starts communicating with the peripheral. (Note that when the app is suspended it's awoken to handle BLE enable/disable and connect/disconnect events without the background-fetch mechanism so it doesn't need any time to try to connect.) And when the app is launched by background fetch, it has a total of 15 seconds from the time our JavaScript code starts executing to initialize and connect.

One thing to keep in mind is iOS does not extend JavaScript timers to compensate for suspension time so if the app scheduled something to happen in 5 minutes and then was suspended for 15, that scheduled code will start running immediately when the app is awoken (probably before the callbackFn).

Background fetch is unpredictable in terms of timing. You can sometimes trigger it by enabling and disabling airplane mode.

randdusing, 2017-06-28:
Are you recommending implementing the willRestoreState? It's probably not too difficult to iterate through the peripherals and set them up. It should probably be opt in since it's quite a change. I'm also not sure how to tell the user what devices were restored. Perhaps the init callback can return this information?

alexhisen, 2017-06-28:
Yes, I think willRestoreState needs to be implemented for a clean solution. But unless you want to break all existing apps, I think, the plugin needs to pretend nothing is connected to let the app go through the normal bootstrap process but just internally recognize the connection and skip doing connects/discovers.

randdusing, 2017-06-29:
It won't break the apps as long as it's opt in!

I don't have much time to devote to this project anymore, so I'm still deciding how to proceed. I would like to set aside a weekend and try to knock out some of the bigger issues. Finding some other contributors would help as well.

sunnymtl, 2018-06-29:
Thank you @alexhisen for your detailed introduction regarding iOS State Preservation and Restoration, it's very helpful! Currently I'm able to achieve it on iOS 10.3 with ionic 3, but after I upgrade to iOS 11.4, the ionic app cannot be launched in background. Have you tried it on iOS 11.4? 

Compared with the log from iOS 10.3 (normal relaunch), I found that in iOS 11.4 it stopped before this line:
0x14fe6b4b0 - FrameLoader::checkLoadCompleteForThisFrame: Finished frame load (frame = 0x1028bd300, main = 1)

Thanks for any help!

alexhisen, 2018-07-02:
What kind of background launch is not happening for you in 11.4 - BLE-triggered or background-fetch-triggered?

sunnymtl, 2018-07-03:
@alexhisen  The app is not really launched in background with the BLE-triggered event, I don't use background-fetch plugin in this app. 
From the log I can see that the app is actually launched when receive a BLE notification, but it stopped at some point (about 2s after receive the first notification, see attache log). If the BLE device send a 2nd notification at that point, the launch will continue until it finish, otherwise it just stuck there.  Which means it need two BLE signals which are 2s apart to trigger a full background launch, this is not going to happen when the device go out of range and come back again. I don't know if it's related to ionic or just the iOS system? Have you seen any issue like this with iOS 11.4? Thanks a lot!
[BLELOG1.zip](https://github.com/randdusing/cordova-plugin-bluetoothle/files/2157452/BLELOG1.zip)

sunnymtl, 2018-07-03:
Thank you Alex for your time! What platform are you using? is it ionic?
Another question is how do you deal with the similar problem for Android system? i.e, if Android kills the app in background, there is no State Preservation and Restoration mechanism for Android. The background-fetch plugin doesn't seem to work when the system kills the app (it works when the app is terminated manually), or maybe the android background service is the only way to deal with this? I guess it's not that easy to realize this for ionic app...
Thank you for the help!

alexhisen, 2018-07-03:
We don't use ionic, though it doesn't seem like that would have any effect. We also use WKWebView currently, though previously we used UIWebView and I think it worked fine on iOS 11 too.

On Android, our hack solution is launching app every 10 minutes using https://github.com/Haggus/cordova-plugin-wakeuptimer (and detecting launch by wakeup at app startup and sending ourselves into background when launched that way).

sunnymtl, 2018-07-03:
Great, thanks a lot Alex! I will give it a try!

alexhisen, 2018-07-14:
Actually, maybe I am wrong about 1 packet. Our device sends packets every second until our app essentially acknowledges.

sunnymtl, 2018-07-14:
In our case, we found that the double signal to successfully launch the app might only happen in older phone model (before iphone 6), models after that the app can be launched with 1 signal. Thanks for the reminder anyways!

adelin-mcbsoft, 2019-11-12:
Hi,

The implementation of the restoration process got me a bit confused... I think it's incomplete.

In [Core Bluetooth (Apple Docs)](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW5) is mentioned that when the app is waken up (after being previously killed), the system invokes the `centralManager:willRestoreState:` method _(as a callback)_, so you can do the restore process on your side, using the _restoreKeys_ provided as a dictionary to the second parameter (see [docs](https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/1518819-centralmanager) here).

Basically, it tells you which connection has triggered the app. (based on the `requestKey`), so you can call `bluetoothle.initialize()` with the initial (same) `requestKey` so you can continue your previous connection, without having to reinitialize it again.
(...at least that's my understanding of it, I'm not an iOS Developer).

However, the current implementation of `centralManager:willRestoreState:` is empty and seems not to be linked in any way with the Javascript part, so basically even if the system invokes the mentioned method, you have no correspondent callback in Javascript which you could use to know your app. has been woken by the system in background to handle and continue your previously initialized connection.

Code (link [here](https://github.com/randdusing/cordova-plugin-bluetoothle/blob/master/src/ios/BluetoothLePlugin.m#L2084)):
```
- (void)centralManager:(CBCentralManager *)central willRestoreState:(NSDictionary *)dict {
  //Needed to support background mode
}
```

Is there any chance to have that callback implemented in the near future?
Many thanks,
A.

adelin-mcbsoft, 2019-11-15:
After digging further, found out that there's more to it:

Let's assume a previously connected BLE device comes back in range _(after being out of it)_, the phone sees that and iOS connects back to the BLE device, while restarting your app. in background to perform the BLE tasks according to its last calls.

I've tested the situation even without having the callback implemented, by putting the code at the very-beginning in a test app _(but still - **the callback is highly needed**, so you can know when the app. was started by the BLE-system call)_.

So, when the app. starts in background:
If the BLE connection is NOT initialized with the previously used `restoreKey`, the remote BLE device disconnects, which I find it to be normal.

But if you indeed initialize BLE with the previously used `restoreKey`, the device stays connected.
However, `bluetoothle.wasConnected` and `bluetoothle.isConnected` both return false.

Calling `bluetoothle.connect` gets the connection back while the remote BLE device doesn't perform a disconnect/new connection, but using the one that was previously started by the iOS, that triggered the background app. launch.

The saddest part is that you have to perform loop through all the connections and try to connect to each, 'cause as you don't have the callback info, you don't know which device came back in range. Scanning doesn't help as the device is technically connected even before the app. started and doesn't advertise.

The expected behavior would be to have proper connection restored when calling `bluetoothle.initialize()` using the `restoreKey`... (and if possible, even having the callbacks of the previously pending BLE calls triggered -- basically, restoring the entire `bluetoothle` object to its previous state).

I'll be happy to provide even more info if needed to have this functionality properly set; just ping me.

Thanks,
Regards,
A.

TobiasFP, 2020-01-27:
Hi @adelin-mcbsoft 
Thank you very much for the walk through.
Do you have any code showing this behaviour?
I myself have developed an app which works flawlessly (well, almost) on Android, but which has difficulties on IOS. I have implemented the restoreKey, gatt services, etc. and I have tried everything to make this work with IOS and the restorekey, but there might just be something I am missing?
Currently, I am scanning on a loop, but I have also tried just calling startscan.
Whenever a device is discovered, onconnected gets called and a chain of event leads to a notify starting on the app, waiting until a ble device notifies me of changes, and it seams 75% of the time, everything works fine, even in background, but the last 25% simply stops the app from responding in background.

I guess my question is, is there some special logic that i need to implement for the restorekey to function properly? Is my only option to loop though a list of devices that I have known has been connected?
Does the restorekey only function when the device+iPhone has been paired via the settings menu in IOS?

adelin-mcbsoft, 2020-02-02:
Hi @TobiasFP .
Sorry for my late reply.
So, from my experience with this library:
`restoreKey` is just an identifier that iOS needs in order to recognize the app. from other existing apps that use BLE... is not really of any use in your app's logic, but **IT MUST BE THERE** in order to have your app restarted (and recalled) from background.
Most importantly, the value of this identifier must not be changed over the app's lifetime.
And it has nothing to do with the BT Settings in iOS menu... I never touched them.

Regarding the steps:
I never re-use the `scan()` feature in order to reconnect to a previously connected device.
I do the following:
1. `scan()` _[once, only the first time per device - let's say at "pairing" - though it isn't really a pairing]_
2. `connect()`
3. `OnDisconnect` _[event]_ (considering the device is off or no longer in range), calling `reconnect()` .

This way, on the hardware-layer the phone continuously looks for the previously connected device.
For the cases when the app. was killed in the meantime and has to be restarted in background by the OS, you'll have to loop through the previously connected devices list and call `connect()` for each, and handle the one that returns the _success_ callback, in order to get the connection back - *as in the current state - the library does not let you know that the app has been started by BLE, nor tell you which BLE device has triggered it.* (too pity... this is needed so much).

Anyway, as you don't have this callback, if you fail to call back `connect()` within seconds in order to restore the connection handle, it will time-out and will no longer reconnect again.
(so yeah, to answer your second question, this is the only way to do it currently... until @randdusing helps us with the _so needed_ callback implementation 😁 ).

Hope this answers all of your questions.
If I can help you with anything else, let me know.

All the best,
A.

TobiasFP, 2020-02-02:
Thank you so much for the reply, awesome info.
Very kind of you.

For anybody else reading this,I have managed to get another flow working 99% of the time, but only on ios 13.

I have added the key in info.plist,  and added the restore key.
I then have a scanner function which "startscanifnotscanning" which is called every five seconds. I know, it is not the most sound logic, but it seems to be 100100% stable on Android, and 100% stable on ios in foreground. 

Besides that, everything is pretty much like are stated in the readme of this plugin. 
I am still having problems on ios 12 though, which is not scanning in background at all.
I might try your set up, even though i actually went away from that just that a couple months ago.

adelin-mcbsoft, 2020-02-03:
IMHO, your approach is not efficient at all (and I don't recommend it) since having the scan active all the time drains the battery of the user, and also - it won't fully work in background under iOS (since you'll have only the 16-bit UUIDs available and most probably won't pass iTunes policies).

Try implementing the solution above, you'll have better results.

TobiasFP, 2020-02-03:
I couldn't agree more with you, Adelin :)