randdusing/cordova-plugin-bluetoothle #436: Scanning Does not return anything in iOS Background Mode

PrakrutiVolansys, 2017-06-28:
I am trying to scan my BLE device in background mode with ServiceUUID specified in startScan() method. But, iOS can not scan it in background.

My ionic info
global packages:

    @ionic/cli-utils : 1.4.0
    Cordova CLI      : 7.0.1 
    Ionic CLI        : 3.4.0

local packages:

    @ionic/app-scripts              : 1.3.7
    @ionic/cli-plugin-cordova       : 1.4.0
    @ionic/cli-plugin-ionic-angular : 1.3.1
    Cordova Platforms               : android 6.2.3 ios 4.4.0
    Ionic Framework                 : ionic-angular 3.1.0

System:

    Node       : v7.10.0
    OS         : macOS Sierra
    Xcode      : Xcode 8.3.3 Build version 8E3004b 
    ios-deploy : 1.9.1 
    ios-sim    : 5.0.13 
    npm        : 5.0.3 

Cordova plugins installed on my system : 

cordova-plugin-background-mode 0.7.2 "BackgroundMode"
cordova-plugin-background-mode-bluetooth-central 1.0.0 "Bluetooth Central Background Mode"
cordova-plugin-bluetoothle 4.4.0 "Bluetooth LE"
cordova-plugin-mauron85-background-geolocation 2.2.5 "CDVBackgroundGeolocation"

randdusing, 2017-06-29:
And it does work in foreground?

PrakrutiVolansys, 2017-06-30:
Yes, It works in foreground.

randdusing, 2017-07-02:
Are you using a 'default' service UUID? I don't think custom UUIDs are allowed from what I remember....

TobiasFP, 2020-01-10:
Hi @randdusing 
Thank you so much for an amazing plugin, using Ionic 4 and EVERYTHING is working perfectly on Android. IOS, I have the same problem as above, and with a custom UUID, do you remember where it is stated that custom uuids are not allowed in background mode?

orhan-swe, 2020-01-27:
@TobiasFP 
Hi, we have the same issue, ios will not scan for our service when app is in the background mode. Did you manage to find a solution? The uuid we have is a one we created for our service: "FFEC". I am not sure what the difference is between a custom and a default service UUID. Do you know?

TobiasFP, 2020-01-27:
Hi Orhan.
Yes, a custom UUID is a non-standard UUID, and if you create one, you should make sure to create a unique UUID, by creating a UUID V4 using https://www.uuidgenerator.net/. Do not just use "FFEC".

I am able to sometimes scan for the UUID, but make sure to use the "restoreKey" in the initialize function.
The functionality works sometimes, but it seams close to impossible to ensure stability.
Are you pairing your devices when doing the bluetooth connection?

TobiasFP, 2020-01-27:
@orhan-swe Btw. a non-custom uuid is a uuid with a certain specific functionality, you can see the specifications here:
https://www.bluetooth.com/specifications/assigned-numbers/service-discovery/

orhan-swe, 2020-01-29:
@TobiasFP 

Thanks for the answer. I will check if we can change the UUID. 
We are not pairing only scanning for the device.

Though we did manage to get background scanning to work on ios also. I think the issue was that we did not have UIBackgroundModes enabled in info.plist , once we added "bluetooth-central" it started to work. This is how it looks:
```
<key>UIBackgroundModes</key>
	<array>
		<string>bluetooth-central</string>
	</array>
```

The background scan may sometimes take more than 5 seconds before it finds our device, this is not good but I am not sure we can do anything about that.
Also we have to restart the scan after each the device is found, so that it can be detected again.

We are not setting restoreKey, I will check to see if it does change anything.