randdusing/cordova-plugin-bluetoothle #437: Initialize returns wrong MAC Address and devices are not discoverable while advertizing

jbesan, 2017-06-30:
Dear all,

I am testing this plugin but I can't get my devices visible no matter if startAdvertizing initiated without error. Digging a little I realized that getAdapterInfo returns a weird MAC address (different from the one I get in the phone's settings): 02:00:00:00:00:00 and I guess the two are linked.

Environment:
- AngularJS 1.4.3, Cordova 5.2.2, Ionic 1.2.4
- Android 7 on 3 different devices: Sony Z4 tablet, Samsung GS6 and Xiaomi mi5

I try to play between the devices (no beacon or Rasberry) using them either as "master" or "slave".
The getAdapterInfo returns:
`{"address":"02:00:00:00:00:00","name":"Xperia Z4 Tablet","isInitialized":true,"isEnabled":true,"isScanning":false,"isDiscoverable":false}`

The addService returns:
`{"service":"00001234-0000-1000-8000-00805f9b34fb","status":"serviceAdded"}`

The startAdvertizing returns success.

On the orther device, the startScan works well and I see the different bluetooth devices nearby (my computer, headset etc) but not the ones advertizing with this module. I even forced the device to be discoverable (with another plugin) which makes the device visible from the bluetooth settings but not through the app.

Side note: I am quite new at using cordova plugins and may have improperly installed the plugin but since I get some results while scanning it is not completely wrong either.

Any ideas ?

Thanks in advance,

Jack

randdusing, 2017-07-02:
Can you try using the example in the Angular wrapper? https://github.com/randdusing/ng-cordova-bluetoothle

If it does work, it's likely an issue with your Javascript code. If it doesn't work, it's likely an issue with the plugin / hardware.

jbesan, 2017-07-04:
Thanks for the quick reply.
Unfortunately it is not that easy for me to test the example since I am using an IDE (Appery.io that makes not that easy to add "custom" pages).

However I pushed further my investigations and it seems I am strucgling becaus this google limitation:
https://www.socketmobile.com/es/about-us/news-room/featured-article/featured-article/2016/02/10/the-case-of-the-missing-bluetooth-address

I don't have 5.x Android device so I can't test on them but do you think it could be the reason?

Thanks in advance,

randdusing, 2017-07-04:
Ahh, that's unfortunate, but I can understand why Android did that. iOS does something similar. Did you ever try scanning for the advertising device and then connecting? Rather than trying to initiate the connection directly via the MAC address?