randdusing/cordova-plugin-bluetoothle #441: Android 6.0 status 133 on connection.

DaniloCouto, 2017-07-04:
I've been testing on various devices and I've got in my hands an Moto G3 with android 6.0.
I try to connect to my slave with this moto g3, but it always return "connection rejected" on the connect method.

`$cordovaBluetoothLE.connect({ address: slave.address, autoConnect: false, timeout: 10000 }).then(.........
`
So I look inside android's log to see what status it's returning.
```
07-04 16:25:32.014  1222  1337 D BluetoothGatt: connect() - device: AB:C4:9F:AA:AF:5C, auto: false
07-04 16:25:32.014  1222  1337 D BluetoothGatt: registerApp()
07-04 16:25:32.014  1222  1337 D BluetoothGatt: registerApp() - UUID=691b41ed-d8c3-42e2-acce-7d386ace54ec
07-04 16:25:32.015  8104  8114 D BtGatt.GattService: registerClient() - UUID=691b41ed-d8c3-42e2-acce-7d386ace54ec
07-04 16:25:32.016  8104  8126 D BtGatt.GattService: onClientRegistered() - UUID=691b41ed-d8c3-42e2-acce-7d386ace54ec, clientIf=5
07-04 16:25:32.016  1222  1358 D BluetoothGatt: onClientRegistered() - status=0 clientIf=5
07-04 16:25:32.017  8104  8115 D BtGatt.GattService: clientConnect() - address=AB:C4:9F:AA:AF:5C, isDirect=true
07-04 16:25:32.017  8104  8126 D bt_btif_config: btif_get_address_type: Device [ab:c4:9f:aa:af:5c] address type 0
07-04 16:25:32.017  8104  8126 D bt_btif_config: btif_get_device_type: Device [ab:c4:9f:aa:af:5c] type 2
07-04 16:25:32.017  8104  8133 E bt_btif : Connection open failure
07-04 16:25:32.018  8104  8126 D BtGatt.GattService: client onConnected() - clientIf=5, connId=0, address=AB:C4:9F:AA:AF:5C
07-04 16:25:32.018  1222  1233 D BluetoothGatt: onClientConnectionState() - status=133 clientIf=5 device=AB:C4:9F:AA:AF:5C
```

So it's returning status 133, that it doesn't appear on android's documentation.
I tried harder to find some answers and I found those :
[This GATT android code, showing the meaning of this status 133 (85 on Hex)](https://android.googlesource.com/platform/external/bluetooth/bluedroid/+/master/stack/include/gatt_api.h)
[This Android 6.0 Issue on AutoConnect = True](https://issuetracker.google.com/issues/36995652)
[This question, with some guy recommend some lib as answer](https://stackoverflow.com/questions/27280402/every-connection-request-is-being-treated-as-direct-connect-request-android-bl)

According to these links, it's an Android 6.0 issue and the possible workaround for this android bug would be turning off the AutoConnect...
But it IS false.
Anyone have any suggestion to workaround?

randdusing, 2017-07-16:
Issues like this are why I start to strongly dislike Bluetooth...

rogerclarkmelbourne, 2017-07-28:
@DaniloCouto 
Can you post some example code to demonstrate this ?

ejbp, 2017-10-30:
Did you solve this issue? Can you please share how? Thanks.