randdusing/cordova-plugin-bluetoothle #444: Read characteristic fails when connection parameter updates are ongoing (Android)

acutetech, 2017-07-14:
I have a test app that scans, connects, discovers, reads two characteristics (battery level and device info serial number) then disconnects, as fast as possible. I notice that in some circumstances the battery level read fails ("Unable to read on return") - though the serial number read works OK (possibly because it is cached rather than needing to be read from the peripheral). In the fail case the peripheral (based on CSR1010) does not see the request to read the battery level characteristic.

The  failure appears to be correlated with connection parameter updates sent by the Android devices, immediately after the connection is complete. These occur differently on my two Android devices - for example when bonding takes place, or on the first connection after disabling/enabling Bluetooth on the device.

A (not very elegant) workaround is to insert a delay between the .discover() and .read() calls. For me 500ms works but 300ms is too short, though results may vary.

This is perhaps not a serious problem, in the sense that the app can detect the issue and work around it. But (a) it is worth reporting it here and (b) perhaps there is a better workaround than the delay and (c) is it possible to fix this in the plugin?

acutetech, 2017-07-14:
Hmmm. The delay between  .discover() and .read() calls works on my Asus tablet but not my Samsung phone. With the phone it seems that the bonding is initiated only when the phone attempts to read the battery characteristic. The bonding works but the characteristic read does not reach the phone, and the plugin returns "Unable to read on return".

In this case it seems that it is not the ongoing connection parameter update that breaks the read (there is none), but the bonding process. So the issue is a more general "read characteristic fails when phone is doing other stuff".

randdusing, 2017-07-16:
You are waiting for the discover callback to complete before calling read, right? Anyway to get a more detailed error log?

aaviles24, 2018-03-02:
acutetech were you able to find the cause of the problem?