randdusing/cordova-plugin-bluetoothle #445: Hyphens must be removed from 128-bit UUIDs for .startScan() (Android)

acutetech, 2017-07-14:
Previously it appeared that I had to send an empty services[] array as a parameter in .startScan(), since if I submitted a single 128-bit UUID then no devices would be found.

I have just discovered it is necessary to remove the hyphens in the UUID string. This does not work:
MED_SERVICE_UUID = "55feb600-0030-11e7-93ae-92361f002671";
services: [MED_SERVICE_UUID]
This does work:
MED_SERVICE_UUID = "55feb600-0030-11e7-93ae-92361f002671";
var service = MED_SERVICE_UUID.replace(/-/g, '');   // find all '-' and change to ''
services:[service]

This puzzles me, as the plugin's java seems to use strings with hyphens, like these (lines 292)
private final UUID clientConfigurationDescriptorUuid = UUID.fromString("00002902-0000-1000-8000-00805F9B34FB");
and the use of baseUuidStart and baseUuidEnd (which assume hyphens).

Also, discover() returns 128-bit UUID strings with the hyphens. 

At minimum, please update the documentation.

randdusing, 2017-07-16:
Does this issue occur on both iOS and Android? I may need to try getting the UUID from the original string and also the 'cleaned' string. It may also have something to do with it being a custom UUID rather than a standard UUID like the one for clientConfigurationDescriptionUUID

mdailor, 2020-09-30:
I have found this to still be the case on Android 10. 

services["7fafe390-2d8b-55ae-e74b-e15600100c7d"] finds no devices.
services["7fafe3902d8b55aee74be15600100c7d"] finds my device.

On iOS, passing a UUID with the hyphens removed will generate an error in BluetoothLePlugin.m as it's not a valid UUID, so not recommended for iOS. :)