randdusing/cordova-plugin-bluetoothle #447: How get manufacturerData on Android

Bill-Niz, 2017-07-17:
Hi,
How can i get the manufacturerData when scanning with android?

randdusing, 2017-07-17:
On Android, the advertisement property will contain a base64 encoded string of bytes. You'll need to convert to bytes and then process the array to determine the manufacturer data. Unfortunately Android doesn't make this easy like iOS does.

Bill-Niz, 2017-07-18:
`ScanRecord scanRecord = scanResult.getScanRecord();
SparseArray<byte[]> manufacturerData = scanRecord.getManufacturerSpecificData();
for(int i = 0; i < manufacturerData .size(); i++){
    int manufacturerId = manufacturerData.keyAt(i);
}`

Bill-Niz, 2017-07-18:
How the bytes are parsed ? I mean in with position the manufacturerData its located ?

randdusing, 2017-07-30:
This could probably help with parsing the advertisement packet: https://github.com/tessel/bleadvertise But how to parse the manufacturerData is going to depend on the manufacturer.

rogerclarkmelbourne, 2017-07-30:
I don't think you need to bother with loading a whole parser just to look for the manufacturer data.

This is a modified version of the parser I use.
and it should give the start index in the advertising array.

Note if you want the start of the actual data, you need to add 2, and the length of the data is the first byte -1

```
parseAdvertising:function(advData)
{
	var s;
	var currentPos=0;
	var advBytes = bluetoothle.encodedStringToBytes(advData)
	console.log(advBytes);
	while(advBytes[currentPos]!=0 && currentPos<advBytes.length)
	{
		if  (advBytes[currentPos+1]==0xff)
		{
					console.log("Found manufacturer data at position "+i);
		}	
		currentPos+=advBytes[currentPos]+1;// Move to next item in advertising.(+1 is because the 
	}
}
```