randdusing/cordova-plugin-bluetoothle #448: Subscribe callbacks not executed when subscribing to a protected characteristic - Android

acutetech, 2017-07-24:
I attempt to subscribe to notifications from a characteristic that has protection, while unbonded, I see this:
(a) The phone and peripheral execute a bonding operation. As I understand it, this is expected: the link is not encrypted until attempting to access a protected resource.
(b) Neither of the .subscribe() callbacks (subscribeSuccess, subscribeError) are executed. 

I would expect the subscribeError call should be executed.

The next time I attempt the subscribe it works (the onSuccess callback executes), because the right level of encryption is in place.

In contrast, when attempting to read a protected characteristic, the phone and peripheral also perform a bonding operation, but the .read() call returns an error "Unable to read on return".

randdusing, 2017-07-30:
Do you ever call bond? I would try calling bond before reading or subscribing

acutetech, 2017-07-31:
I do indeed use this workaround: call isbonded() and if unbonded, call bond(), before accessing the protected resources. So there is a simple workaround. I was just pointing out the anomaly - perhaps either to be fixed or to be documented.