randdusing/cordova-plugin-bluetoothle #449: Android startScan services filtering does not seem to work

rogerclarkmelbourne, 2017-07-28:
I built a test app, and it works fine with 

`bluetoothle.startScan(startScanSuccess, handleError, { services: [] });`

However if I attempt to filer for a specific 16 bit service ID,  e.g. 1234

`bluetoothle.startScan(startScanSuccess, handleError, { services: ["1234"] });
`
I still get devices the same list of devices found, including ones which have a service which is not "1234"

I know Android has some problems with 128 bit service UUID's but I thought 16 bit service UUID scan filtering was OK

I've tried this on both Android 7 and Android 7 devices and I get the same result

randdusing, 2017-07-30:
Did you mean Android 6 and 7? Is the service an 'official' one or custom? Android may not like custom 16 bit UUIDs.

rogerclarkmelbourne, 2017-07-30:
I'm running Lineage OS for both Android 6 and 7, because none of my devices can be officially be update to Android 6 or newer.

Re: 16 bit service ID's

Its a custom 16 bit UUID, which I was using for testing, but I can easily change it for an official Bluetooth.org 16 bit ID number.

However I know Android works with custom 16 bit UUID's as a lot of Chinese companies don't bother to register the ID's they use in their devices with Bluetooth.org (probably because of the cost of doing so)

And their Apps still work OK.

I'll need to do some more testing, but I did some testing with the ble-central cordova plugin and it was able to filter for 16 bit UUID's, however it may be doing this in the plugin code rather than passing the filter to Android's BLE API

rogerclarkmelbourne, 2017-07-31:
Seems to work OK on Android 6, so I'm going to close this until I can definitively know if this is an Android 7 or a hardware issue

closed 2017-07-31 by rogerclarkmelbourne