randdusing/cordova-plugin-bluetoothle #450: Android 7 - disconnect() and close() do not disconnect

rogerclarkmelbourne, 2017-07-29:
Testing on Android 7 (Lineage OS), I've found that neither disconnect() or close() seem to close the connection to the BLE device

I'm using Nordic's nRF52 Development kit, running the Heart Rate Service example, with added debugging to log when connections are made and disconnected and on Android 6 I see 
"Disconnected, reason 19" logged from the dev kit, which is the same as if I use something like LightBlue on the iPad.

However on Android 7, the Development kit does not log that its getting disconnected, and if re-run my test app, it does not discover the dev kit again, until I reset it, so its definitely not advertising.

I guess I'll need to see if I can find whats different in Android 7's BLE handling which is preventing it from disconnecting

randdusing, 2017-07-30:
Very interested in your tests. Other people are having similar issues.

rogerclarkmelbourne, 2017-07-30:
@randdusing 

I've done some more tests but its hard to know if this is a Android 7 or Lineage OS issue, or potentially a hardware issue on my Samsung J500F

I have Android 7 (Lineage) also running on Moto G2 2014 phone, but BLE does not seem to be working at all on that device, so its probably an problem in Lineage OS.

I've tested another BLE plugin on my Samsung (running Android 7 Lineage) and I get the same issue with disconnection, which probably means that there is an issue with the phone or Lineage on that phone.

I've posted my APK to some friends and asked them to test this, but its not that easy to test unless you have a BLE device where you can log from the connection and disconnection functions.

I'll let you know if I get any feedback from them.

And... Of course Android 8 should hit the streets soon, so theoretically I should be able to try it on my Moto G2 (2014) hardware, as Lineage have said they will support Android 8 on the Moto G2 when its released

acutetech, 2017-07-31:
My app uses disconnect() then close(). This works as I would expect on two Android 6 devices but not on an old Android 5 phone - my peripheral does not see the disconnect (it eventually times out and disconnects itself). On the Android 5 phone I can use the Nordic Semi nRF application to connect and disconnect to the same peripheral - in this case the peripheral does see the disconnect (which shows there is not an inherent problem with the phone and its stack) - with the same 19 status reported above - I thing this means the BLE connection is terminated by the Central.

I then dusted off an old app that uses cordova-plugin-ble from Evothings. It has a close() but no disconnect(). On the Android 5 phone the close() is seen by the peripheral. 

So it may be instructive to see how cordova-plugin-ble close() code differs from cordova-plugin-bluetoothle

rogerclarkmelbourne, 2017-07-31:
@acutetech 

I'm glad it's not just me that has an issue with connections not being closed ;-)

I think it's entirely possible that many people are not noticing this problem, because it's not obvious from the Android side of the communication and most BLE devices are probably programmed to disconnect after a period of inactivity.
( but Nordic's demos don't seem to implement a timeout, hence I noticed that I was unable to reconnect after I reloaded my Android App, because the Nordic demo thought it was still connected to the previous instance of the App.

I will download the Evothings BLE plugin and see if that works with Android 7 (Limeage OS) on my Samsung J5

rogerclarkmelbourne, 2017-08-01:
@acutetech 

I just retested my BLE device and the Nordic nRF Connect App does not seem to disconnect either when I use my Samsung J500F which is running Android 7

My other Android 7 phone seems to have a fault on its Bluetooth, as its no longer detecting anything at all. But this is more likely to be an issue with Lineage OS on old Moto G2 hardware than Android 7.

Unfortunately I don't know anyone with a phone running Android 7 :-( who could test this for me on their phone

acutetech, 2017-08-01:
Hmmm... it might be more complex than I reported yesterday. If I restart my Android 5 phone and run my own app then when my app does the disconnect() and close() operations my peripheral does disconnect. However if I then open the Nordic Semi nRF Connect app I get the problem I reported yesterday: when my app closes the BLE connection my peripheral does not see this - the BLE connection remains up. It seems that the nRF app is keeping the connection up. Note that I do NOT ask nRF Connect to connect to my peripheral, and the autoconnect selection is off. But I do see in the nRF log that it is connecting to my peripheral. When I click the nRF DISCONNECT link, nRF also reports "Disconnected" but the link remains up until my peripheral times out and shuts it down.

I returned to one of my Android 6 phones and see similar (but not identical) behaviour: nRF Connect connects to my peripheral even if I don't ask it to, and keeps the BLE connection open. But my peripheral sees a remote disconnect followed immediately by a new connect. Clicking DISCONNECT in the nRF app does close the connection.

A colleague also working on BLE confirms that nRF keeps BLE connections open - she is not using this plugin, so it is an issue related to Android and nRF. In general, what is there to prevent (or control) two apps from interacting with the same peripheral? 

Today I have also seen a new phenomenon (same app and Android 5 phone). You need to understand that my peripheral starts advertising immediately after the BLE link is closed. Sometimes my app closes the connection but the peripheral sees a connection immediately followed by a disconnect by the central, and this loops continuously (Until, I think, I disable BLE on the phone). So the phone connects and disconnects continuously in a short loop. It is not my app that is doing this.

A final note - on the Android 6 setup only, sometimes when the phone disconnects my peripheral see the connection close with status 0x3d "MIC failure" rather than 0x13 "Remote". 

What a mess!

A question for Roger - does your problem go away if you uninstall nRF connect?

rogerclarkmelbourne, 2017-08-02:
@acutetech 

I didn't have nRF Connect (or nRF Toolbox) installed on my Android 7 phone.
I've only tested the nRF app disconnecting on Android 7, I don't normally use their App that much, I use some other iOS apps for debugging, like LightBLue and Pally BLE Scanner.

Its interesting that the nRF Connect app does not seem to disconnect for you either.
If this is consistent across multiple phones, its probably worth raising an issue on github and / or posting to the Nordic Devzone.

If I get time I'll try the nRF app on some other phones and tablets as I have A 4.4 A 6 and A 7

I was also considering reloading my Samsung J5, which is currently running A 7, back to its latest available manufacturer firmware, in case the issue was Lineage OS (Android)

philips77, 2017-08-08:
Hi, here's the author of nRF Connect. 
The app doesn't do anything special, or automatic. It just logs and does based on the events it get from the system (via onConnectionStateChanged or a broadcast). It may connect when gets ACTION_CONNECTED (?) and there is no tab open, but I'm not sure. Also, it connects if gets onConnectionStateChanged with status connected to the server (server must be enabled and option in Settings must be On). 
However, we encountered a similar/same issue with SDK 12+, where after disconnecting a device would connect again, reported multiple times to NUS profile. I think it may be related to quick notifications. When you disconnect, but few ms after that a notification is received, a phone behaves as if it was a Connect request. If you have notifications enabled and going on, try disabling then before disconnecting. It's kind a workaround, and may not help. In nRF Connect I observed multiple times a notification after disconnect in the log, that's the reason of my suspicion. Also, could you post here logs from nrf connect, the part where is reconnects.

acutetech, 2018-09-11:
I had this problem again (not using Cordova this time). It turns out that the Nordic Semiconductor nRF Connect app (useful for debugging) keeps the BLE connection open. There is an explanation here:

https://devzone.nordicsemi.com/f/nordic-q-a/38422/android-disconnect-and-close-do-not-disconnect