randdusing/cordova-plugin-bluetoothle #452: new function

fjlogrono, 2017-08-16:
Could you add a function to know if the device is compatible with BLE? Would be useful would not it?

FEATURE_BLUETOOTH_LE

rogerclarkmelbourne, 2017-08-17:
Can you explain what you mean ?

phy25, 2017-08-17:
I guess he's talking about adding a shortcut to run `getPackageManager().
hasSystemFeature(PackageManager.FEATURE_BLUETOOTH_LE)` in Android (
https://stackoverflow.com/questions/18873905/how-to-find-out-if-android-device-supports-bluetooth-low-energy
).

rogerclarkmelbourne, 2017-08-17:
I wonder is calling getAdaptorInfo can be used instead, but I can't test this as I don't have any devices without BLE

fjlogrono, 2017-08-17:
Yes, that's phy25. a shortcut to run `getPackageManager()

randdusing, 2017-09-10:
This should be easy to add. I may be able to do it within the next week or so, but PRs are welcome.