randdusing/cordova-plugin-bluetoothle #453: Android turn off Bluetooth freeze the app

sunnymtl, 2017-08-17:
Thanks Rand for the great plugin!

I have a small problem: whenever I disable the Bluetooth, the App running on Android phone freezes, but the same App on iOS still works.  I have tried on Android 5 and 6, both behave the same. 

If there is a subscription when Bluetooth is disabled, it just throw out isDisconnected message from subscribeError callback. And then I cannot run ANY bluethoothle method (no reaction), and all buttons in the App are frozen. And I cannot close the App with back button on phone. If I unsubscribe before the Bluetooth is disabled, the App freezes as well.

All other ways to disconnect the link works fine, except manually disable Bluetooth. 

Which callback should be called when Bluetooth is disabled? I know if the BLE device is disconnected or out of range will be caught from status => disconnected in connect() method, but it doesn't look like the case when Bluetooth is disabled. 

Thanks a lot for any help!

sunnymtl, 2017-08-25:
Has anyone run into the same problem?

randdusing, 2017-09-10:
The plugin shouldn't cause buttons in the app to freeze. Can you try using the example app in the angular wrapper and see if the same behavior exists?

sunnymtl, 2017-09-11:
Thank you Rand! I will try the angular wrapper.