randdusing/cordova-plugin-bluetoothle #454: App crash in Windows 10

rifathaqueamit, 2017-08-30:
Hello.

I am having a problem with windows Javascript application. During the running of the application, if I disable the bluetooth and then re-enable it again, and after that, if the subscribe() function is called then the app crashes.

It is happening after calling the subscribe() function for the above circumstance only. Can you provide any clue why this is happening ?

randdusing, 2017-09-10:
Any stack trace or error log? Can't provide much input without that.

rifathaqueamit, 2017-09-14:
I solved it by myself. In the function getService(...), when a service is found by fromIdAsync(), it pushes the service object into the cachedService array. After I turn bluetooth off and then on, this service object gets lost that is already in the cachedService. If the getService(...) is called again, it returns success as the serviceId and deviceId is still inside the cachedService array, although the service itself does not work at that time. So I commented out 
'    for (var i = 0; i < cachedServices.length; i++) {
      var service = cachedServices[i];
      if (service.deviceId == deviceId && service.serviceId == serviceId) {
        successCallback(service.deviceService);
        return;
      }
    }'
inside the getService(..) function and it solved the crashing problem.