randdusing/cordova-plugin-bluetoothle #457: Too slow

MarioAlmanzaAmaya, 2017-09-26:
is there any way to reduce the time you run discover since it's taking too long

Thanks in advance

randdusing, 2017-10-01:
Which operating system? On Android, everything is discovered. On iOS, you can choose which services and characteristics to actually discover.

MarioAlmanzaAmaya, 2017-10-02:
Thanks for replying, it only happens to me in android, but only in version 7.0 of my moto g5 plus, in the others I work well.

swetha-thoomoju, 2017-10-03:
I have also seen the delay in discovering the services in some android devices One plus and Lenovo looks like issue does not depend on the OS version

swetha-thoomoju, 2017-10-03:
@randdusing its just a guess do we have to discover services similar to iOS, is there any option for that?

I see different **BluetoothGatt** logs in with different devices

**Delay(~8s) scenario (Lenovo)**
10-03 07:23:43.516 D/BluetoothGatt(22313): discoverServices()
10-03 07:23:51.321 D/BluetoothGatt(22313): onGetService()
.
.
.
10-03 07:23:51.326 D/BluetoothGatt(22313): onGetCharacteristic()
.
.
.
10-03 07:23:51.391 D/BluetoothGatt(22313): onGetDescriptor()
.
.
.
10-03 07:23:51.446 D/BluetoothGatt(22313): onSearchComplete()

**In no delay scenario (MOTO G3)**
10-03 16:55:39.328 D/BluetoothGatt( 1706): discoverServices() 
10-03 16:55:40.805 D/BluetoothGatt( 1706): onSearchComplete() Status=0

randdusing, 2017-11-09:
There's no way around at is as far as I know.

MarioAlmanzaAmaya, 2017-11-09:
According to what I have tried, in the smartphone with android 7 it takes 7 to 9 seconds to turn on the bluetooth. I do not know if it also appears on native android