randdusing/cordova-plugin-bluetoothle #460: Any issues on iOS 11?

qpwr, 2017-10-09:
This is not an issue, but a question: are there any known issues with this plugin in iOS 11 (e.g. connection drops etc.)? I am currently using the evothings plugin which has been randomly discontinued and would like to transition over to a new plugin because the evothings stuff doesn't work properly on iOS 11.

Thanks for any input on this.

francovp, 2017-10-11:
I can't find beacons on iOS with evothings iBeacon fork, so I think is a iOS 11 incompatibility with the plugin, because on iOS 10.3.3 works fine.

scottcarter87, 2017-11-02:
I've recently noticed that the plugin is no longer capable of enabling bluetooth on the users behalf but I know for a fact that this worked in older versions of iOS I just don't know what version broke this functionality.

randdusing, 2017-11-09:
@scottcarter87 I'm reproducing that on iOS 11 as well. I don't remember exactly what it said previously, but I think this behavior may be expected now...

jcviau, 2018-01-09:
We have issues with iOS 11.1+ and bluetooth 5 devices (iPhone 8, X). We arent able to establish a connection with our peripheral since the connect calls never complete (promise not resolved). It looks like an issue Apple introduced in iOS 11.1. Not sure what exactly yet...

The plugin works well with other devices (iOs & Android) and was working with iOS 11.0 and bluetooth 5 devices.

skifaster, 2019-04-29:
Anyone have luck with iPhoneX yet? We are getting stuck on connect calls as well. Any other cordova suggestions for BLE that are working with iPhoneX?