randdusing/cordova-plugin-bluetoothle #461: Scan does not work for Samsung Galaxy S8

TheJacobYoung, 2017-10-10:
I have two mobile devices - a Samsung Galaxy S5 and a Samsung Galaxy S8. BLE devices show up on a scan with my S5 but not my S8. I am using the same app for both. Are you aware of this issue or is there already a  fix?

francovp, 2017-10-11:
Doesn't work in my Moto G5 Plus with Android 7.1.2 either. What version of Android do you have? I think is a problem with Android 7+

TheJacobYoung, 2017-10-12:
I have Android version 7.0

As an alternative, I have found that this other ble plugin works: https://github.com/evothings/cordova-ble

francovp, 2017-10-16:
Unfortunately that plugin doesn't have a function like bluetoothle.initialize or bluetoothle.enable :(

randdusing, 2017-11-09:
Main difference is this plugin takes advantage of the new scan functionality that was introduced in Lollipop. I could add a variable to force old usage, or you may be able to fix the issue by tweaking some of the scan parameters.

Hypher, 2017-11-12:
@francovp I don't think the bug you cite is in anything related to the actual problem. See #465 I got the same problem but another plugin works so it's not an Android bug (or there is a workaround).

sunny-mittal, 2018-01-17:
@TheJacobYoung can you verify location permissions have been requested and granted? I'm running Android 7 on an S8 and have no problems. I do think the plugin should be better about reporting errors (specifically, when location permissions haven't been granted) but that's the only thing I can think of. Let me know if you are still having an issues and I can help to debug further.