randdusing/cordova-plugin-bluetoothle #464: Initialize status always returns disabled (iOS 11.2)

hannibaldinski, 2017-11-02:
I'm experiencing weird behaviour of the window.bluetoothle.initialize() function on iOS11.2 Public Beta. 

When I trigger the function, the callback object .status is always returns 'disabled'. 
Even though BL is enabled and Allows New Connections.

After toggling Bluetooth the function does return the .status 'enabled'.

iPhone 7
iOS 11.2 - Public Beta
Cordova v7.1.0
Ionic v3.1.2
cordova-plugin-bluetoothle v4.4.2

hannibaldinski, 2017-11-02:
It magically disappeared... I am very confused.

Anyone else with similar problem?

randdusing, 2017-11-09:
I can't install the betas anymore, so I can't verify myself. Does it happen on the same phone if downgraded to an earlier version? Assuming downgrade is even possible...

hannibaldinski, 2017-11-12:
Not in a position to downgrade, though I got report from a iPhone 5s user (iOS 11.0.2) who seems to experience the same issue.

I did found a iPhone 5s myself. I'll start testing it if I can reproduce it.

closed 2019-03-08 by hannibaldinski