randdusing/cordova-plugin-bluetoothle #465: Samsung A5 2017 / Android 7.0 : nothing is working

Hypher, 2017-11-12:
```"request": true,``` does not request for BT enabling dialog, it just calls the callback with ```res.status = 'disabled'```.

```startScan``` callback once with ```res.status == 'scanStarted'``` but never call it again with a found device, although I have one in range and other apps find it.

For instance https://github.com/don/cordova-plugin-ble-central have no problem with both issues.

hannibaldinski, 2017-11-16:
I've the same issue on the same phone but with Android 6.0.1.

I've around 15 active beacons around and only 3 could be found during scan.

All other device models return all 15 beacons during scan.

alfonso8969, 2018-01-03:
I have the same problem with the same phone and with the same version of Android (7.0).
Even passing an array with the generic and specific parameters of different devices.

lovemonkey, 2019-10-10:
You may should use 'bluetoothle.requestPermission' and 'bluetoothle.requestLocation' first, then you will be find it is OK!