randdusing/cordova-plugin-bluetoothle #474: cannot connect in android & Phonegap Build

Electrobyte2, 2018-03-03:
i try in phonegap and this plugin but is imposible to connect. Initialize correctly the plugin but always show error in the connect metod. this 's my code:

bluetoothle.initialize(initializeSuccess, initializeError);
}
    function initializeSuccess(obj)
       {
     bluetoothle.connect(function connectSuccess() {
alert("connect Success!");
}, function writeError(err) {
 alert("Error:" + err.code + "" + err.message)
}, { "address": "DC:2C:26:06:45:65" });
  
i'm use Phonegap Wild and in my config.xml use:
<plugin name="cordova-plugin-bluetooth-serial" spec="0.4.5" />

i appended this permissions:

 <preference name="android.permission.BLUETOOTH" />
<preference name="android.permission.BLUETOOTH_ADMIN" />   
   

Thanks guys
 I

randdusing, 2018-03-03:
Did you request permission within the app?

Electrobyte2, 2018-03-04:
Hi, how do I do that?

randdusing, 2018-03-04:
https://github.com/randdusing/cordova-plugin-bluetoothle#haspermission
Actually this may not be scanning related at all since you're only trying to connect, but chances are the address isn't correct. Take a look at the documentation for hasPermission, requestPermission and startScan. Try to scan for the device and see if you can find it before trying to connect to it.