randdusing/cordova-plugin-bluetoothle #476: Cannot connect to ble in Android 7.1.0

paulrodz, 2018-03-28:
Connect method throws error "connection failed" and if tried to connect again throws error already connected use reconnect or closed.  However, I checked it is not connected to ble device

LocutusOfPenguin, 2019-01-07:
I have the same problem with Android 8.0.0
Is there any hope on solving? Thanks.

sdelrue, 2019-10-07:
I fixed this issue by placing a timeout inbetween connects. The error was caused by trying to connect too fast to the same address.

nejchi31, 2019-11-05:
how do you do that?

sdelrue, 2019-11-05:
something like this:
```
let connected = false;
let intervalId = setInterval(() => {
    if (!connected) {
        bluetoothle.connect(connectSuccess => {
            connected = true
            clearInterval(intervalId);
            //device connected
        }, connectError => {

        }, params);
    }
}, 5000);
```