randdusing/cordova-plugin-bluetoothle #477: iOS: initialize promise resolve not called

rastafan, 2018-03-31:
Hi everyone,

we encountered an issue when using this plugin on iOS. We are using Ionic V1.
Basically, we put this code at the very start of the app (we tried both in the app.js and at the first page enter):

```
new Promise(function (resolve) {
	bluetoothle.initialize(resolve, {
		request: true,
		statusReceiver: false
	});
}).then(function (result) {
	if (result.status === "enabled") {
		log("Bluetooth is enabled.");
		log(result);
	} else {
		log("Bluetooth is not enabled:", "status");
		log(result, "status");
	}
}, handleError);

```

but it never enters the success function, nor the error handler.
It appears to be working fine in Android, the only difference is that we call the same code but in the success function of the `bluetoothle.requestPermission` function.

Any ideas? Did we miss something?

randdusing, 2018-03-31:
Are you waiting to call initialize until Cordova is ready?

rastafan, 2018-04-01:
Yes, everything is started inside the ionicPlatform.ready() callback.
I forgot to say that even though the success and error methods are not triggered, successive scans and read/write operations are most of the time successful, and sometimes it says "Bluetooth0 not enabled".
We tested it on iPhone X with iOS 11.3

randdusing, 2018-04-03:
Does the example app in the angular wrapped cause the same issue?

rastafan, 2018-04-04:
The angular wrapped version seems to work fine, thanks, we totally missed that. 
We will probably stick to that.

andersborgabiro, 2018-06-17:
I have the same issue in iOS. Android works fine.

initialize doesn't generate any callback. I've tried both with and without a promise (not sure why a promise is needed), and with and without restoreKey. This is after deviceready and bluetoothle is initiallized (startScan can be called, but fails). There are also plist settings for location and Bluetooth. There are no error messages.

Oddly, it worked for a while, and then very much not, so I suspected conflict with other apps, and removed all apps that might use BLE in the background. No difference.

I checked the Angular example and couldn't see anything obvious that differed.

This is the code ("initBLE" is shown, but nothing else) :

```
function initBLE() {
    log("initBLE", true);

    var promise = new Promise(function(resolve) {
        bluetoothle.initialize(resolve, {
            request: true,
            statusReceiver: false,
            restoreKey: "abirotesteddystoneurl"
        });
    });

    promise.then(initializeSuccess, handleError);
}

function initializeSuccess(result) {
    log(result, true);

    if (result.status === "enabled") {
        log("Bluetooth is enabled", true);
    }
    else {
        log("Bluetooth is not enabled", true);
    }
}

function handleError(result) {
    log(result, true);

    if (result.error && result.message) {
        log("Error on " + result.error + ": " + result.message, true);
    }
    else {
        log(result, true);
    }
}
```

andersborgabiro, 2018-06-19:
If I remove the options for initialize it works, so by default:

```
request: false,
statusReceiver: false
restoreKey (none)
```

aroncal, 2018-10-17:
I had a similar issue when porting in the Ionic v1 app some commits from Android to IOS, including a cordova-plugin-bluetoothle upgrade from 2.4.0 to 4.4.4 and the corresponding modifications in calls to library's methods such as `initialize`. 
I found out that the problem came from not having upgraded the plugin inside the IOS platform, as I had done previously inside the Android platform, because `npm install` upgrades the plugins in `node_modules` honoring `package.json`, but it doesn't upgrade them inside `platforms/ios`. You must `ionic cordova plugin rm cordova-plugin-bluetoothle` and then `ionic cordova plugin add cordova-plugin-bluetoothle`.

psanty, 2019-03-19:
Our team has the same problem, updating the package didn't work either. At the moment, the only solution we've found is very dirty and involves bypassing the initialize callback, instead resolving the promise after a timeout. The initialization, as we found, does happen, the only thing that doesnt work is the callback.

JesperBalslev, 2020-02-03:
@randdusing This is still a problem, and i suspect this is also the reason why there are issues with the connect callback with autoConnect.

hschultz1960, 2020-05-05:
Confirmed from here too ... 

This works:

```
bluetoothle.initialize(resolve, {
    "request": true
    });

```
but adding `statusReceiver `and `restoreKey `causes the callback to never happen.

SamuelMichelRiad, 2021-07-19:
Any update on this, other than omitting the statusReceiver and restoreKey? This is still an issue in version 6.5.1, and the workaround by @hschultz1960 still works.