randdusing/cordova-plugin-bluetoothle #478: Subscribe Error: Service not Found

rfbezerra, 2018-04-19:
Hi,

Im trying to communicate with a custom equipment without success.
My needs is for running my app in Android 4.r (API 19), so, I'm using old 2.4.0 version.

I can connect, discover services, disconnect.
My problem is when subscribe to a characteristic: "Service not found"

Here is the discover results:
```json
{
  "address":"XX:XX:XX:XX:XX:XX",
  "status":"discovered",
  "services":[{
    "characteristics":[{
      "descriptors":[],
      "characteristicUuid":"2a00",
      "properties":{"read":true}
    },{
      "descriptors":[],
      "characteristicUuid":"2a01",
      "properties":{"read":true}
    },{
      "descriptors":[],
      "characteristicUuid":"2a02",
      "properties":{"write":true, "read":true}
    },{
      "descriptors":[],
      "characteristicUuid":"2a03",
      "properties":{"write":true, "read":true}
    },{
      "descriptors":[],
      "characteristicUuid":"2a04",
      "properties":{"read":true}
    }],
    "serviceUuid":"1800"
  },{
    "characteristics":[{
      "descriptors":[{"descriptorUuid":"2902"}],
      "characteristicUuid":"2a05",
      "properties":{"indicate":true}
    }],
    "serviceUuid":"1801"
  },{
    "characteristics":[{
      "descriptors":[{"descriptorUuid":"2902"},{"descriptorUuid":"2901"}],
      "characteristicUuid":"ffe1",
      "properties":{
        "writeWithoutResponse":true,
        "read":true,
        "notify":true
      }
    }],
    "serviceUuid":"ffe0"
  }],
  "name":"PeripheralName"
}
```

And error occurr with:
```javascript
bluetoothle.subscribe(resolve, reject, {
  address: 'XX:XX:XX:XX:XX:XX',
  service: 'ffe0',
  characteristic: 'ffe1',
})
```
```json
{"message":"Service not found","error":"service","address":"XX:XX:XX:XX:XX:XX","name":"PeripheralName"}
```

Did I miss something?

Thanks

chicag0, 2018-05-01:
When I subscribe neither the success nor failure functions fire

marcui13, 2023-02-08:
Hi, how are you?
I have the same problem, were you able to solve that?

sdetweil, 2023-06-30:
you must discover to update data used by the library.. until you do, you cannot read/write, or connect really