randdusing/cordova-plugin-bluetoothle #479: Background timer for scan or reconnect

sunnymtl, 2018-04-26:
I need to connect with a BLE device all the time. There is no problem when the app is in foreground. 
In case the device is disconnected in the background, we have to use a timer to schedule the scan and reconnect. Now the problem is that the timer may not work when the app is in Background (both Android and iOS), so Scan or Reconnect will never be fired in this case. I have tried katzer/cordova-plugin-background-mode but it drains battery and it's said will be rejected from Apple store. Is there a good way to arrange the Scan or Reconnect when the app is in Background? Thanks a lot!

ferreirabraga, 2018-06-06:
I'm stuck the same issue here. Anyone could help?

sunnymtl, 2018-06-14:
@ferreirabraga Just use reconnect() after receive disconnected call back, no timer needed on both iOS and Android! I guess reconnect() can run forever in background without consume power (from my test).

closed 2018-06-15 by sunnymtl