randdusing/cordova-plugin-bluetoothle #480: Bluetoothle is not defined

Trauck, 2018-05-03:
Hi, I added this plugin to connect in bluetooth to a device. So I ran the installation "cordova plugin add cordova-plugin-bluetoothle", then I try to use it in my javascript related to my html page. But when I build on my device, nothing happens like when there is a js error.
Here but version of android and cordova:
android 6.4.0,
cordova 8.0.0.
I call bluetoothle like this in my scripts:
[image]
Thank you.

randdusing, 2018-05-03:
You're probably calling plugin code before the plugin is actually ready. Are you waiting for the cordova ready event?

Trauck, 2018-05-04:
Yes sure, i make the same thing with "$(document).ready(function(){ ... ".

randdusing, 2018-05-05:
That's when jquery is safe to use, not cordova plugins. I forgot the event name off the top of my head, but check out the example in the Angular wrapper

Trauck, 2018-05-14:
It's good, finally everything worked very well I forgot the addition of a script (cordova.js).

I still have a new question. When I press a button on my application I launch my bluetooth connection, my bluetooth is activated very well, but then I would like to show the list of possible connections. 
How to do this ? I try with the Scan but when I run it, nothing appears.

DonOzOn, 2022-04-16:
> It's good, finally everything worked very well I forgot the addition of a script (cordova.js).
> 
> I still have a new question. When I press a button on my application I launch my bluetooth connection, my bluetooth is activated very well, but then I would like to show the list of possible connections. How to do this ? I try with the Scan but when I run it, nothing appears.
> 
> And what is the function that turns off the bluetooth of the phone?

HI Trauck , I face to same issue , can you show me how you addition of a script (cordova.js)? Thanks so much