randdusing/cordova-plugin-bluetoothle #485: Are there plans to fix the issues with Indicate?

tempestwf, 2018-06-02:
We recently switched to your plugin from the official one for ionic for our app development and we love it so far. Great job creating so many features and giving the users to control over how every scenario is handled.

We are wondering if you have any plans to fix the issues you mention your documentation about Indication subscription functionality.

The docs state "Indication type subscription hasn't been well".

The device my company manufactures we are planning to switch to using indicate method in the coming months, so I was interested to know if there are plans to address the issues referred to in the docs?

Thanks for making a great plugin.

Will

tempestwf, 2018-06-07:
Thanks for the info. Are you available for contract work to improve the plugin? I am trying to get a sense of our options to keep improving the plugin going forward.

All the best

Will

randdusing, 2018-06-08:
Hi Will, unfortunately I'm not available at the time to do contract work.

tempestwf, 2018-06-09:
Thanks, Randdusing. Please let me know if you get freed up in the future.

Hope you have a great weekend.

Will

lucagessi, 2022-05-02:
Hi.
I would like to fix this issue, at least in Android. Do you have some more information?