randdusing/cordova-plugin-bluetoothle #487: Unable to connect on Windows

MalikDrako, 2018-06-06:
I am able to scan for bluetooth devices, but when I try to connect on Windows, there is a native popup saying "Connecting to <device>" followed by "Connection failed".

Stepping through the plugin JS, the error is logged from BluetoothLePlugin.js#L265, but the DevicePairingResultStatus code is 19: An unknown failure occurred.

My connection code is pretty simple:
```
app.bt_connect = function(address) {
	var params = { address: address };
	bluetoothle.connect(onConnect, onConnectError, params);
}
function onConnectError(result) {
	console.error(result);
}
```

MalikDrako, 2018-06-21:
It turns out the device I am trying to connect to doesn't support pairing. Newer versions of Windows 10 support communicating without pairing, how would I modify the plugin to allow this?

randdusing, 2018-06-23:
Not experienced with the windows side of BLE at all...

closed 2018-12-25 by randdusing

randdusing, 2018-12-25:
Please reopen if @TheBosZ 's recent changes didn't fix this.

PR #511 (2018-12-25): Rewrite Windows 10 support
This PR is to bring Windows 10 UWP support up to about the same level as Android and iPhone.

I did what I could to make the experience the same between the platforms.

We're using this plugin in an app that's targeting Android, iOS and Windows. With this PR, the app works as expected across the three.

The major missing functionality is advertisement data. I'm still working on that, but right now it appears as though the end-user has to jump through a lot of hoops to get it working so it may not be viable.

This closes #487, closes #491, closes #501, closes #508, an [... cut]