randdusing/cordova-plugin-bluetoothle #488: Plug in not working when packaged into an APK

tempestwf, 2018-06-17:
Hi

The plugin is working perfectly when running:

ionic cordova run android --device

To deploy to a device from my computer.

However, if we run:

ionic cordova build --release android

(then signing and optimize it)

To make an APK, and then when we install the APK the plugin doesn't work. The bluetooth device is not being detected.

We will have more intel on this issue as soon as we are putting error reporting on our deployed app, but for starters has anyone encountered this before?

Thanks very much.

Will

tempestwf, 2018-06-17:
Updating with more exact info about the issue.

tempestwf, 2018-06-21:
We were able to solve this problem. It turns out that https://github.com/randdusing/cordova-plugin-bluetoothle#requestpermission must be called when the application is packaged into an APK. It's not an issue when using ionic cordova run android, however.

Not sure why there is a difference, but that is the fix.

randdusing, 2018-06-23:
You may have previously allowed permission when using ionic run, and then when you deployed the release apk, it may require permissions to be requested.

tempestwf, 2018-08-20:
Thanks Rand

The problem doesn't seem to be happening any more. We made some major
changes to our implementation of the plugin, and this and some other issues
went away.

All the best

Will