randdusing/cordova-plugin-bluetoothle #490: Android 8 not sending timely Bluetooth events when in Doze/ App Standby mode

benhtn, 2018-06-20:
I am attempting to create an Ionic v1 app the purpose of which is to maintain a connection to a bluetooth LE device and notify the user if it disconnects (along with trying to reconnect etc) using the Angular wrapped version of this plugin.

However during testing on Android 8.0 I've found that the appropriate callback isn't being called upon disconnection in anything like a timely fashion (it seems to wait until the next OS maintenance window, as explained here: https://developer.android.com/training/monitoring-device-state/doze-standby) when the device is in a sleep/ doze state.  See example below:

```
    $cordovaBluetoothLE.connect({
        address: address,
        timeout: timeout
    }).then(
        null, 
        function(e) {
            return onConnectError(e);
        }, 
        function(result) { // this does not happen in a timely way when device is dozing
            if(result.status === "connected") {
                onConnected(result);
            } else {
                onDisconnected(result);
            }
        }
    );
```

I have tried the following (on a Moto G6 running Android 8.0):
 - I have set our app to run in background mode using these plugins:
   - https://github.com/katzer/cordova-plugin-background-mode
   - https://github.com/randdusing/cordova-plugin-background-mode-bluetooth-central

 - I have manually turned off battery optimizations specifically for this app via Settings > Battery > Battery Optimization

randdusing, 2018-06-23:
Same behavior across Android devices and versions? I don't remember running into many issues with Android and background modes, but it's been a while since I've done any BLE work.

bramk, 2019-08-21:
Any new insights on this?

I have a strange but similar case with the same setup as @benhtn. In addition, after the BLE device is connected, I subscribe to a service that sends me data at about 50hz. This worked fine before and still does while the app is in the foreground. However, on devices running Android 8+ I now no longer receive the `disconnected` notification while I still do reveive data once the app is in "doze mode".

More detail:
1. When the app is in the foreground I receive the `subscribedResult` notifications and `disconnected` notifications on the respective callbacks as expected.

2. When I move the app into the background this still works for about 5 minutes. I assume at this point the app enters "doze mode".

3. In "doze mode" I still receive the `subscribedResult` notifications as long as the BLE device is connected, but when it disconnects I do not receive the `disconnected` notification.

4. Once I move the app in "doze mode" back into the foreground I immediately do receive the pending `disconnected` notification.

App setup:

- ioniv v1
- cordova-android@8.0.0 (targetSdkVersion 28)
- cordova-plugin-bluetoothle@4.5.5
- cordova-plugin-background-mode-bluetooth-central@1.0.0
- cordova-plugin-background-mode@https://github.com/equestic/cordova-plugin-background-mode
- app excluded from battery optimization

bramk, 2019-08-23:
Ok, my problem was a result of incorrectly use of the cordova-plugin-background-mode plugin. It turns out that calling `disableWebViewOptimizations()` works if you call it **after** the app goes into the background. So as a result my webview went into "doze" mode and maybe that was what happend to @benhtn too.

So, through it is still  a riddle to me why I did receive `subscribedResult` notifications and not `disconnected` notifications, my problem is now solved.