randdusing/cordova-plugin-bluetoothle #492: Bluetooth Developer Portal hyperlink results in 404

AlexChambers, 2018-06-22:
Link: https://developer.bluetooth.org/gatt/services/Pages/ServicesHome.aspx

The link to navigate to a list of "out of the box" UUIDs is broken and leads to a 404. Not sure if the page requires authentication/authorization to view, but would expect to see another error if that was the case.

randdusing, 2018-06-23:
Hmm, working for me. Perhaps a proxy issue? Anyone else experiencing this issue?

It redirects to https://www.bluetooth.com/specifications/gatt/services

Does that link work for you? I'll probably change it either way

AlexChambers, 2018-06-24:
It very well could be a proxy issue. I'm able to access the link on my home WiFi, but was having trouble using work WiFi.

randdusing, 2018-06-28:
Updated the link!

closed 2018-06-28 by randdusing