randdusing/cordova-plugin-bluetoothle #493: Error on startScan

RitterD, 2018-06-25:


phy25, 2018-06-25:
Maybe only location service is required to turn on, rather than GPS? On newer Android, before using Bluetooth users need to turn on location service and grant location access to the app.

RitterD, 2018-06-25:
Hi phy,
thank you for a quick response. When I say GPS, I mean the location service, sorry....
Is there an option to turn on the location service by initialization or even better enable it for the app in the background?
When I use my Bluetooth to connetct with my car or some speakers, I never turned on the location service....
It was realy annoying when the code dosn't work and all you have to do is enable the location service...

phy25, 2018-06-25:
@RitterD Please read https://github.com/randdusing/cordova-plugin-bluetoothle#haspermission . It seems that the permission needs to be granted by user interaction.

eht22, 2021-01-06:
May you please help me every time I am implementing your code I am getting error Class Not Found. when i use method start any function of this bluetoothLE i am getting error class not found

eht22, 2021-01-06:
@RitterD if you know pls help

RitterD, 2021-01-13:
Hi eht22,
its hard to tell after 2y.... as I remember, this was in the very start of my project. check out the documentation and follow it carefully and start here:
https://github.com/randdusing/cordova-plugin-bluetoothle#sample-discover-and-interact-with-bluetooth-le-devices

hopefully it helps. keep trying the plugin works