randdusing/cordova-plugin-bluetoothle #494: Sensor only response with 4 UUIDS

RitterD, 2018-06-26:
Hello!

I'm currently programming a smartphone app for my bachelor thesis where I want to connect the smartphone with a sensor to read some tool holder data.

Every time I connetct to the sensor I only finde 4 characteristics, but the sensor supposed to response with more, apparently...(like 2B, 2C) After 4 characteristics an error also occurse and the bluetoothle.read never fires...

[image]

Do I need advertising first bevor the sensor response with more UUIDS maybe and what happens in this function where "**characterisitcs**.forEach" starts?

`
function addService(address, serviceUuid, characteristics) {

    console.log('Adding service ' + serviceUuid + '; characteristics:');
    console.log(characteristics);

    var readSequence = Promise.resolve();

    var wrapperDiv = document.createElement("div");
    wrapperDiv.className = "service-wrapper";

    var serviceDiv = document.createElement("div");
    serviceDiv.className = "service";
    serviceDiv.textContent = uuids[serviceUuid] || serviceUuid;
    wrapperDiv.appendChild(serviceDiv);

    characteristics.forEach(function (characteristic) {

        var characteristicDiv = document.createElement("div");
        characteristicDiv.className = "characteristic";

        var characteristicNameSpan = document.createElement("span");
        characteristicNameSpan.textContent = (uuids[characteristic.uuid] || characteristic.uuid) + ":";
        characteristicDiv.appendChild(characteristicNameSpan);

        characteristicDiv.appendChild(document.createElement("br"));

        var characteristicValueSpan = document.createElement("span");
        characteristicValueSpan.id = serviceUuid + "." + characteristic.uuid;
        characteristicValueSpan.style.color = "blue";
        characteristicDiv.appendChild(characteristicValueSpan);

        wrapperDiv.appendChild(characteristicDiv);

        readSequence = readSequence.then(function () {

            return new Promise(function (resolve, reject) {

                bluetoothle.read(resolve, reject,
                    { address: address, service: serviceUuid, characteristic: characteristic.uuid });

            }).then(readSuccess, handleError);

        });
    });

    document.getElementById("services").appendChild(wrapperDiv);

    return readSequence;
}
`

Thank you very much in advance =)

randdusing, 2018-06-28:
Not really following, this looks like a Javascript issue. You're trying to access uuids, which is undefined

RitterD, 2018-06-28:
yes and I already found uuids but only 4 there apparently supposed to be more....

you know, its weird that I can finde 4 uuids and after that en error occurs and says: uuids are not defined...

RitterD, 2018-06-29:
Hi @randdusing,

I found the cause of my problem. You mentioned in your example that: "That file does not appear in this example, but the values in it come from this page on the **Bluetooth web site: Characteristics.**"
The uuids I couldn't finde, and caused that error, are custom uuids which are propably not in "The array of uuid values that is used in this example...."

It is realy important for me to finde custom uuids also because they contain the data (temperatur, run time,..) which I urgenty need. Is there any option to include more uuids. You said they come from a help function which includes a list of the characteristics, where can I finde that list and how can I add more uuids? I need this for my Thesis...

Thank you very much in advance!

randdusing, 2018-07-01:
uuids isn't defined anywhere. This is a pure javascript issue and doesn't appear related to the plugin

RitterD, 2018-07-05:
adding an array with the name uuids fixed the problem. =)
I'm able to finde all charactersitcs now thank you very much!!

now I need to figure out how I read some values constantly to track there changes do you have any recommendations?..
but this might be another topic, the uuids problem is solved, you can close the topic

thanks @randdusing

closed 2018-07-10 by RitterD