randdusing/cordova-plugin-bluetoothle #496: Connect and Disconnect isn't stable on Android

tretrau1014, 2018-07-05:
When I connect to my BLE device, at the first time, everything is good until I disconnect -> close and connect again, sometimes I cannot connect to, sometimes it took a long time to connect.

tretrau1014, 2018-07-18:
Can you show me your code? Last time I set a small timeout between connections but I still got the same issue.

sunnymtl, 2018-07-18:
What BLE device are you using? I'm using nRf51822, so my setting is:
#define MIN_CONN_INTERVAL        MSEC_TO_UNITS(60, UNIT_1_25_MS) 
#define MAX_CONN_INTERVAL       MSEC_TO_UNITS(90, UNIT_1_25_MS)
#define SLAVE_LATENCY                 12

This works for me for a stable connection on both Android and iOS.

tretrau1014, 2018-07-18:
I used [cordova-plugin-bluetoothle](https://www.npmjs.com/package/cordova-plugin-bluetoothle) for my Ionic 3 application, 
Do u have any Ionic code for that? I followed the official documents but it still got issue

sunnymtl, 2018-07-18:
If you can connect, then your code should be fine. I'm following the documents as well. Your problem probably relates to your device.

ducnoivu, 2018-07-19:
But It works well on IOS device, that's problem.

kgillespieatmosphere, 2018-07-20:
On Android do not disconnect just close, it's an odd issue. Also to speed up connection and discovery continue to scanning while you are connecting and discovery a BLE device on android. This can increase the speed by 2 times in some cases.

There isn't much people are going to be able to do to help your issue with Android. it's just kind of stinks for anything running before version 7.0.0 of Android.