randdusing/cordova-plugin-bluetoothle #504: How can I have the right Mac address on getAdapterInfo method? Always returning "02:00:00:00:00:00"

elysiopires, 2018-10-04:
After install the plugin I tried to get the mac address of my current device by executing:
bluetoothle.getAdapterInfo(function(ren){console.log(ren);},function(ren){console.log(ren);});

and I received as a response at all devices:
address: "02:00:00:00:00:00"
isDiscoverable: false
isEnabled: true
isInitialized: true
isScanning: false
name: "Galaxy Tab A (2017)"

I just install the plugin... Am I missing something to get the right address?

Thx in advance!

Best,
Elysio

Maximvdw, 2018-11-08:
Android removed the ability to obtain the device MAC address in Marshmallow or above.
https://developer.android.com/about/versions/marshmallow/android-6.0-changes#behavior-hardware-id
The plugin will need to be modified to account for these changes

randdusing, 2018-12-21:
What type of modifications to the plugin are you expecting? There's already support for requesting finer grained location permissions.

adamalexander, 2019-01-07:
I'm facing the same issue and as a result my device isn't discoverable. Any progress?

MrLrx, 2020-03-25:
@randdusing Hello, has this problem been solved? Android 6.0 can't get the real MAC address

MrLrx, 2020-03-25:
@elysiopires Has your problem been solved？

elysiopires, 2020-03-25:
No! After android 6 they implemented security policies that prevent you to
access this information.

Best,
Elysio